gnuton/asuswrt-merlin.ng #312: Ability to switch 2.5G Wan port with 1G Lan port

HDGuy, 2023-01-03:
Installed your latest build on my XT8s 
and it works great. 

One feature on the XT8 that really annoys me is the inability to use the 2.5G Wan port as a LAN port.  I'm sure I'm not the only one who has gigabit internet who would like to use the 2.5G port of the XT8 as a LAN Ethernet when used as your base.

I tried fooling around with dual-wan and even USB Ethernet but none of them will let you do this. 

Do you think it would be possible? With dual Wan you can already use one of the Lan ports as a Wan port which would be adequate for my purposes.

milezzz, 2023-01-19:
Please this, everyone wants this feature and I'm dumbfounded why it isn't stock.

HDGuy, 2023-01-19:
> Please this, everyone wants this feature and I'm dumbfounded why it isn't stock.

When n XT8 is used as an extender the 2.5G port is automatically used as a 2.5G LAN port so I figure it's possible.

psande, 2023-07-15:
+1 for this