gnuton/asuswrt-merlin.ng #313: Asus TUF AX5400 Cannot change icon for devices

bobyang3, 2023-01-03:
**Router Model Affected**
TUF AX5400 

**Firmware Version Affected**
[386.07_2-gnuton1](http://router.asus.com/Advanced_FirmwareUpgrade_Content.asp)

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
I cannot change the icon. (currently, I use Asus app to change it)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Network Map'
2. Click on 'View List'

**Expected behavior**
I can change the icon.

**Screenshots**
[image]

gnuton, 2023-01-04:
I am not sure TUF UI supports it.
You can flash the no_tuf image. 
it should work there.

GaselK, 2023-01-05:
This is no bug. You can not change icon from that list. Icons can change (in ui, tuff or not) from the right list "Client status"
[image]

closed 2023-01-05 by gnuton

bobyang3, 2023-01-06:
> This is no bug. You can not change icon from that list. Icons can change (in ui, tuff or not) from the right list "Client status" [image]

thanks! I guess I need to update it first because I don't have "client status" in 386.

[image]