gnuton/asuswrt-merlin.ng #314: AX92U crash on TrendMicro check

Ick93, 2023-01-04:
**Router Model Affected**
Models: RT-AX92U

**Firmware Version Affected**
388.1 beta2

**Is this bug present in upstream Merlin releases too?**
Not sure

**Describe the bug**
After clicking the check button to update the Trend Micro Software the logs show a memory error, or more often the router crashes and restarts.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to “http://router.asus.com/Advanced_FirmwareUpgrade_Content.asp”
2. Click on “Check” beside “Trend Micro: Signature version”

**Expected behaviour**
router checks for new Trend Micro update and applies it, or states the router is already running the latest version

**Screenshots**

Ick93, 2023-01-04:
System Log of the error

Jan  4 11:57:03 rc_service: httpd 1445:notify_rc start_sig_check
Jan  4 11:57:05 kernel: cfg_server triggered out of memory codition (oom killer not called): gfp_mask=0x201da, order=0, oom_score_adj=0
Jan  4 11:57:05 kernel: CPU: 1 PID: 1761 Comm: cfg_server Tainted: P           O    4.1.51 #2
Jan  4 11:57:05 kernel: Hardware name: Broadcom-v8A (DT)
Jan  4 11:57:05 kernel: Call trace:
Jan  4 11:57:05 kernel: [<ffffffc000087658>] dump_backtrace+0x0/0x150
Jan  4 11:57:05 kernel: [<ffffffc0000877bc>] show_stack+0x14/0x20
Jan  4 11:57:05 kernel: [<ffffffc000527850>] dump_stack+0x90/0xb0
Jan  4 11:57:05 kernel: [<ffffffc0000ffedc>] __out_of_memory.isra.6+0xa4/0x148
Jan  4 11:57:05 kernel: [<ffffffc000100748>] out_of_memory+0x48/0x68
Jan  4 11:57:05 kernel: [<ffffffc0001045e8>] __alloc_pages_nodemask+0x610/0x6a0
Jan  4 11:57:05 kernel: [<ffffffc0000feefc>] filemap_fault+0x15c/0x3c0
Jan  4 11:57:05 kernel: [<ffffffc00011ccdc>] __do_fault+0x34/0x90
Jan  4 11:57:05 kernel: [<ffffffc000121074>] handle_mm_fault+0x9e4/0x1120
Jan  4 11:57:05 kernel: [<ffffffc00008f49c>] do_page_fault+0x10c/0x268
Jan  4 11:57:05 kernel: [<ffffffc000080a9c>] do_mem_abort+0x3c/0x98
Jan  4 11:57:05 kernel: Exception stack(0xffffffc010933e20 to 0xffffffc010933f50)
Jan  4 11:57:05 kernel: 3e20: 00400000 00000000 00000000 00000000 ffffffff ffffffff f69fbfb8 00000000
Jan  4 11:57:05 kernel: 3e40: 00000000 00000000 00000000 00000000 6c749f57 00004f5f 6c73dc07 00004f5f
Jan  4 11:57:05 kernel: 3e60: 000dc440 ffffffc0 1ffe68d0 ffffffc0 00000000 00000000 00000000 00000000
Jan  4 11:57:05 kernel: 3e80: 10933ea0 ffffffc0 000871a8 ffffffc0 00400008 00000000 00000000 00000000
Jan  4 11:57:05 kernel: 3ea0: 00000000 00000000 000844d4 ffffffc0 00400000 00000000 000845f0 ffffffc0
Jan  4 11:57:05 kernel: 3ec0: 00000000 00000000 00000002 00000000 00000000 00000000 00000000 00000000
Jan  4 11:57:05 kernel: 3ee0: 24dd2fb8 00000000 00000106 00000000 00000000 00000000 00000152 00000000
Jan  4 11:57:05 kernel: 3f00: ffd324fa 00000000 000c4c8c 00000000 00000000 00000000 f68aff9c 00000000
Jan  4 11:57:05 kernel: 3f20: f68b04e4 00000000 f68afe28 00000000 f68b0920 00000000 00000000 00000000
Jan  4 11:57:05 kernel: 3f40: 00000000 00000000 00000000 00000000
Jan  4 11:57:05 kernel: Mem-Info:
Jan  4 11:57:05 kernel: active_anon:17129 inactive_anon:461 isolated_anon:0
Jan  4 11:57:05 kernel:  active_file:2 inactive_file:4 isolated_file:0
Jan  4 11:57:05 kernel:  unevictable:0 dirty:0 writeback:0 unstable:0
Jan  4 11:57:05 kernel:  slab_reclaimable:562 slab_unreclaimable:39869
Jan  4 11:57:05 kernel:  mapped:384 shmem:2661 pagetables:630 bounce:0
Jan  4 11:57:05 kernel:  free:10537 free_pcp:75 free_cma:0
Jan  4 11:57:05 kernel: DMA free:42272kB min:20480kB low:25600kB high:30720kB active_anon:68516kB inactive_anon:1844kB active_file:8kB inactive_file:16kB unevictable:0kB isolated(anon):0kB isolated(file):0kB present:436224kB managed:380188kB mlocked:0kB dirty:0kB writeback:0kB mapped:1536kB shmem:10644kB slab_reclaimable:2248kB slab_unreclaimable:158800kB kernel_stack:2672kB pagetables:2520kB unstable:0kB bounce:0kB free_pcp:840kB local_pcp:76kB free_cma:0kB writeback_tmp:0kB pages_scanned:0 all_unrecl
Jan  4 11:57:05 kernel: lowmem_reserve[]: 0 0 0
Jan  4 11:57:05 kernel: DMA: 1780*4kB (UM) 483*8kB (UM) 92*16kB (UM) 141*32kB (UM) 145*64kB (UM) 41*128kB (UM) 6*256kB (UM) 2*512kB (UM) 0*1024kB 0*2048kB 2*4096kB (R) = 42248kB
Jan  4 11:57:05 kernel: 2741 total pagecache pages
Jan  4 11:57:05 kernel: 0 pages in swap cache
Jan  4 11:57:05 kernel: Swap cache stats: add 0, delete 0, find 0/0
Jan  4 11:57:05 kernel: Free swap  = 0kB
Jan  4 11:57:05 kernel: Total swap = 0kB
Jan  4 11:57:05 kernel: 109056 pages RAM
Jan  4 11:57:05 kernel: 0 pages HighMem/MovableOnly
Jan  4 11:57:05 kernel: 14009 pages reserved
Jan  4 11:57:07 BWDPI: fun bitmap = 47f
May  5 01:05:05 crashlog: LOG
May  5 01:05:05 crashlog: <6>[ 1437] 65534  1437      787       79       5       2        0             0 avahi-daemon
May  5 01:05:05 crashlog: <6>[ 1444]     0  1444     2974      497       9       2        0             0 httpds
May  5 01:05:05 crashlog: <6>[ 1445]     0  1445     2871      433      10       2        0             0 httpd
May  5 01:05:05 crashlog: <6>[ 1447]     0  1447     1402       71       6       2        0             0 vis-dcon
May  5 01:05:05 crashlog: <6>[ 1451]     0  1451     1256       61       6       2        0             0 vis-datacollect
May  5 01:05:05 crashlog: <6>[ 1452]     0  1452     1794       84       8       2        0             0 infosvr
May  5 01:05:05 crashlog: <6>[ 1454]     0  1454     1407      541       6       2        0             0 sysstate
May  5 01:05:05 crashlog: <6>[ 1455]     0  1455     3711      330      11       2        0             0 ahs
May  5 01:05:05 crashlog: <6>[ 1456]     0  1456     3149      143      10       2        0             0 watchdog
May  5 01:05:05 crashlog: <6>[ 1457]     0  1457     3149      121       9       2        0             0 check_watchdog
May  5 01:05:05 crashlog: <6>[ 1458]     0  1458     3149      127      10       2        0             0 alt_watchdog
May  5 01:05:05 crashlog: <6>[ 1459]     0  1459     1095      437       5       2        0             0 rstats
May  5 01:05:05 crashlog: <6>[ 1460]     0  1460     3149      126       9       2        0             0 sw_devled
[... cut ...]
May  5 01:05:17 wlceventd: wlceventd_proc_event(486): eth6: Disassoc D8:1C:79:84:F4:82, status: 0, reason: Disassociated because sending station is leaving (or has left) BSS (8)
May  5 01:05:17 wlceventd: wlceventd_proc_event(515): eth7: ReAssoc D8:1C:79:84:F4:82, status: Successful (0)
May  5 01:05:17 kernel: random: nonblocking pool is initialized
May  5 01:05:17 httpd: Succeed to init SSL certificate...80
May  5 01:05:17 httpd: Succeed to init SSL certificate...8443
May  5 01:05:18 BONDING: option disabled
May  5 01:05:19 roamast: ROAMING Start...

gnuton, 2023-01-04:
Tested and I canno reproduce it on beta 1 and 2.
Please reset the firwmare to factory and try again.
Please re-open it if required

closed 2023-01-04 by gnuton