gnuton/asuswrt-merlin.ng #316: ZenWiFi ET8 - RT-AXE95Q Mesh Node not change with main router

quangkieu, 2023-01-06:
**Router Model Affected**
Node Models: ZenWiFi ET8 - RT-AXE95Q
Main Models: ZenWiFi ET8 - RT-AXE95Q
**Firmware Version Affected**
388.1

**Is this bug present in upstream Merlin releases too?**
Yes/No

**Describe the bug**
So when ever I make change in wireless router setting, like WIFI SSID, channels, password of main router, the node is not updated with new information. I did track if Guess Network update correctly before I remove and join the node back to main router

**To Reproduce**
Steps to reproduce the behavior:
1. Change WIFI SSID on Wireless page (mostly because I could easily monitor and check) on main router
2. Hit Apply
3. Monitor on WIFI list to see new name from main router but old name still listed as node router did not change to new WIFI name

**Expected behavior**
After some sync period like 10second, the node should get new WIFI setting (name, channel, password)

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

gnuton, 2023-01-15:
have tou reset the main node to factory after flashing the new firmware?

quangkieu, 2023-01-26:
sorry, i did not able to find down time to do a reset and reconfig yet. Maybe next week