gnuton/asuswrt-merlin.ng #317: TUF-AX5400 DDNS causes disconnection

bobyang3, 2023-01-06:
**Is your feature request related to a problem? Please describe.**
I am not sure if this is a bug or not or only helps are needed so I don't log this as we bug. Currently, *I think*. I got disconnected when the router failed on DDNS update. I don't think it is hardware issue because CPU and memory are good.

[image]

[image]

[image]

gnuton, 2023-01-06:
I do not think DDNS is related to any disconnection.
You got some requet timeout but since you get only one and not multiple and witout resolve or routing errors I do not thnk you got disconnected either

bobyang3, 2023-01-06:
Thank you. Agreed, this is very odd to see failed DDNS will cause this
issue. I always RDP to my work laptop (both computers are at home) and
that's how I found I was not getting ping response from the router and
google DNS. I also ping -t at my work laptop, and I saw it failed to the
router every 5 mins as well.

It was like that for about 30 mins. then I disabled DDNS, and disabled
self-cert. Then, disconnection issues disappeared. I re-enable it again
with the new updated password, so DDNS won't fail. everything is back to
normal. I don't see a non-response ping from the router anymore. Just feel
weird and interesting

gnuton, 2023-01-07:
most probably this is not related to the router itself at all. maybe coming from an upstream router??

closed 2023-01-07 by gnuton