gnuton/asuswrt-merlin.ng #318: TUF AX3000 EVERY 60 SECONDS ROUTER AS A NODE SENDING A REQUEST TO DNS.MSFTNCSI.COM

neil1962, 2023-01-07:
I have tried this with both the current 3881. Beta 3 firmware and the latest stock firmware version 3.0.0.4.388.22237 and they have the identical same problem.  Every 60 seconds a request is being sent to dns.msftncsi.com, this only seems to happen when the router is attached as a Node.  The node is connected by Ethernet.  When returning back to firmware version 3.0.0.4.386.49780 all is good again, i.e. no request being sent every 60 seconds.  

On trying both versions a complete reset was completed, via the WPS button and setup from scratch, no importing of settings.

gnuton, 2023-01-07:
related to https://github.com/gnuton/asuswrt-merlin.ng/issues/182 ?

neil1962, 2023-01-09:
This problem is not there when running the last ASUS stock firmware, only the new 388 firmware is causing this problem, it remains with your latest beta and ASUS latest firmware

gnuton, 2023-01-09:
If I change the log level on 388 you won't see these logs anymore  and it will be just like stock.

neil1962, 2023-01-09:
That's not the problem, the router continues to send these queries, it needs to be stopped so how? The last 386 firmware doesn't have this problem so what has changed?

rlfrank165, 2023-01-09:
I have no idea if this is helpful, but at least on the XT8 in dual WAN (WAN
backup) mode where you can use a USB connected cell phone as a backup WAN
connection, that particular DNS query is the default for simply determining
if the primary WAN is still operational so it can determine if it needs to
switch over to the backup USB connected WAN, and it is set to query every
5  seconds.  Could this query have something to do with something similar
here, simply being used to determine if the primary WAN is operational?  It
is defined under "auto network detection"

Randy Frank

neil1962, 2023-01-09:
Where is the auto network setting?

neil1962, 2023-01-09:
> On the XT8 388 beta 3 it's under WAN and then Dual WAN.
> 
> Good luck.  Hope that may explain this.
> 
> Randy
> 
> 
> > Where is the auto network setting?

Thanks for that, but in my case I don't use a dual WAN

rlfrank165, 2023-01-09:
but my thought is maybe this is also used elsewhere simply as a form of
keep-alive detection for the WAN even without the use of dual WAN.  The
router does need some way of knowing if the WAN is operational.  Might be
worth changing the destination of the DNS site that is DNS-pinged in the
auto network detection part of that tab to see if that changes anything.

When I read your original message the DNS name you mentioned did strike a
bell as I had seen that same name in the auto network detection tab.

Randy

neil1962, 2023-01-10:
The problem here is that the same problem exists in firmawares,3881. Beta 3 firmware and the latest stock firmware version 3.0.0.4.388.22237.  It did not exist in firmware version 3.0.0.4.386.49780.  The network monitoring under administration>system seems to do nothing..Is this setting even pasted down to nodes in an AiMesh setup?  There needs to be a selection of time interval for these requests.

neil1962, 2023-01-10:
> related to #182 ?

Yes similar

gnuton, 2023-01-16:
duplicate of #182

closed 2023-01-16 by gnuton