gnuton/asuswrt-merlin.ng #321: Beta3 Stability (Crash/Reboots)

hyprsnpr, 2023-01-09:
**Router Model Affected**
Models: RT-AX95Q

**Firmware Version Affected**
388.1-gnuton0_beta3

**Is this bug present in upstream Merlin releases too?**
Not sure.

**Describe the bug**
Upgraded to beta3 and router became unstable. It rebooted by itself 3 times within 24 hours. Prior to beta3 I was running beta2 for over a week and it was rock solid. Yesterday I went back to beta2 and it has been very stable since.

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade to beta3.
2. Wait for crash/reboots.

**Expected behavior**
I would expect beta3 to be just as stable as beta2.

felalex, 2023-01-09:
No restarts since upgrade from beta 3 here:
21:45:06 up 1 day, 10:47,  load average: 0.95, 1.11, 1.04

hyprsnpr, 2023-01-09:
> No restarts since upgrade from beta 2 here: 21:45:06 up 1 day, 10:47, load average: 0.95, 1.11, 1.04

Interesting, are you also running on an RT-AX95Q?

felalex, 2023-01-09:
> > No restarts since upgrade from beta 2 here: 21:45:06 up 1 day, 10:47, load average: 0.95, 1.11, 1.04
> 
> Interesting, are you also running on an RT-AX95Q?

yes, maybe the attached SSD for scripts and swap helps:

             total       used       free     shared    buffers     cached
Mem:        510024     431224      78800      14308       5404      50076
-/+ buffers/cache:     375744     134280
Swap:      2097148     102932    1994216

hyprsnpr, 2023-01-09:
> yes, maybe the attached SSD for scripts and swap helps:
>
> ```
>          total       used       free     shared    buffers     cached
> ```
>
> Mem: 510024 431224 78800 14308 5404 50076 -/+ buffers/cache: [...]

Didn't even realize you could attach an SSD. I'll have to look into that. Did you do a reset when flashing beta3 from beta2?

felalex, 2023-01-09:
> Didn't even realize you could attach an SSD. I'll have to look into that. Did you do a reset when flashing beta3 from beta2?
You can run stuff like AdGuard  home on the router (that's what I use it for, among other scripts and swap).
 I didn't manually restart the router , not from the on/off button and not from the web UI.
Whatever the upgrade procedure did by itself, I didn't interfere.

hyprsnpr, 2023-01-09:
Thanks, I updated the same way. Wonder why it's a problem for me and not you.

felalex, 2023-01-09:
> Thanks, I updated the same way. Wonder why it's a problem for me and not you.

Try to enable swap, maybe mom usage is the issue...
Maybe there are logs you can post for the devs?

hyprsnpr, 2023-01-09:
> > Thanks, I updated the same way. Wonder why it's a problem for me and not you.
> 
> Try to enable swap, maybe mem usage is the issue... Maybe there are logs you can post for the devs?

Do you have the swap enabled on the main router and the mesh node?

felalex, 2023-01-09:
> > > Thanks, I updated the same way. Wonder why it's a problem for me and not you.
> > 
> > 
> > Try to enable swap, maybe mem usage is the issue... Maybe there are logs you can post for the devs?
> 
> Do you have the swap enabled on the main router and the mesh node?

Only main unit.

hyprsnpr, 2023-01-09:
Not sure swap will fix it then because my mesh node reboots as well but at different times. However, I have enabled swap and upgraded to beta3 again. I guess we'll see if anything changes.

Thanks for your help.

rlfrank165, 2023-01-10:
Apologize if this msg is a little indicative of my lack of expertise for
this group, but I'm trying to provide as much useful info on these
persistent 388 crashes,  I have a 3 node XT8 (wired ethernet AIMesh).
Yesterday I installed 388 beta 3 after doing a factory reset on the main
router, and things looked good for about a day, and then right at midnight
a few minutes ago, crash...

Once again, this is simply mirroring experiences I had with the official
388 ASUS releases before I switched to Merlin/Gnuton, so I'm highly suspect
of the ASUS GPL code releases and not any Merlin changes, but am still
hoping that the Merlin community may be able to provide ASUS with
information that has made both the official ASUS 388 releases and therefore
the Merlin releases based on that very unreliable.

I have configured recording of syslogs on a connected desktop machine
(using a syslog recorder on that machine) and on the surface this is
working fine.  I get mounds of syslog entries from the many hours the
router is working just fine.

But I have a nasty feeling that the very entries I want, namely those in
the last few seconds before a crash, are getting lost as there is never
anything very interesting recorded moments before a crash.  Question: if
the router has detected a condition that is about to result in a crash,
won't that interrupt the very processes that are responsible for copying
those last second syslog entries from being shipped over the LAN to the
remote syslog recorder?  If the router is dying, won't that prevent any
(fairly complicated) IP-based communications from the router to a remote
LAN connected syslog server? Am I missing something?  How does one capture
those very critical last minute syslog entries that may provide crucial
info on why the router is about to crash and reboot? Are they captured
anywhere in local NVRAM on the router so that one can make sure one is
getting the very last syslog entries that were made just prior to a crash?

Randy Frank

hyprsnpr, 2023-01-10:
Hello Randy, great question on the syslogs,I have been trying to figure that out as well. I too was experiencing the crashes on the official Asus firmware, that's why I switched to Merlin. Evening was great under beta 2. I also suspect the up steam Asus code but hoping the Merlin community and fix it.

I did reinstall beta 3 yesterday afternoon and so far no crashes. I did enable swap but it has not used any swap yet so I suspect that change has not really had an impact. I fear a crash will be coming. Like you said, can't find anything of interest in the logs.

hyprsnpr, 2023-01-10:
Just an update, It's been over 24 hours now with no crashes. Still not confident the issue is fixed but I do see it used some of the swap. So perhaps adding swap did help with this issue. I will continue to monitor.

[image]

JustinFreitas, 2023-01-10:
Maybe there's a memory leak?  Been running for a few days on Beta3 and have seen my free memory gradually be consumed.  This started at 74% memory utilization and just keeps going up (at 80% now).

[image]

[image]

micron888, 2023-01-11:
Tuf ax5400 beta 3
1gb Swap enable
Adguard home 
Crash over night

rlfrank165, 2023-01-11:
Only half joking: what large scale especially real-time software system
doesn't have memory leaks?  That's why I set my router and even desktop to
automatically reboot weekly. But clearly the daily or more frequent crashes
indicate that something out of the ordinary is happening.  But I'm still in
the dark as to how to ensure I'm seeing syslog entries from the moments
before a crash that hopefully might shed some more light on what is going
on here, unless there is some situation where the cause of these crashes
isn't even being trapped.  I recall that someone posted a syslog that
showed a "kernel divide by zero error" just before a crash so clearly it is
possible to get syslog entries that seem relevant to likely causes of
crashes, but nothing I'm seeing from any of remotely captured syslogs show
any sign of why might be causing a kernel level crash.

Randy Frank

gnuton, 2023-01-13:
@rlfrank165 if the router or kernel crashes you  get a lot of logs.
AFAIK the only way you won't get logs, is the app or something is graceously killed or there are hw failures which makes the hardware reboot so the software doesn't detect the problem in time and it has the time to write to disk.

I ahve seen crashes in some closed source modules like trendmicro but usually there is no real harm or rebooting.

jef00, 2023-01-14:
Did you guys do a reset when flashing beta3 from beta2?

rlfrank165, 2023-01-14:
yes I did at least on the main router (full hardware reset, not just the UI
based reset).  I will go back and reset the main router and the two mesh
nodes as well if that is worth trying.

Randy

jef00, 2023-01-14:
that will be worth trying it think.. maybe that will fix it.

hyprsnpr, 2023-01-14:
> that will be worth trying it think.. maybe that will fix it.

I did not. Adding swap seemed to have fixed the issue for me, for now at least.

rlfrank165, 2023-01-14:
Curious, does anyone in the Merlin community have more formal connections
back at ASUS?  Clearly these problems emanate from the ASUS GPL releases,
with the broader ASUS community complaining loudly about the standard 388
releases.  Is ASUS aware of the damage these unstable releases are doing to
their reputation?

Randy

gnuton, 2023-01-15:
@rlfrank165 Asus listen to the feedback sent to them and along with the feedback they  get reports on crashes.
They are really active in fixing stuff, but sometimes it's not easy, requires time and there is no clear info of what's the problem.
As developer this report for instance doesn't tell me how to reproduce the issue, or at least for the time being I am not able to know what's the problem on your side.

Gents, please when you have issues please try to describe all the steps useful for me to test the issue and please attach logs.
Moreover do not go off-topic. otherwise I do read tons of comments I get lost..
As for now the only info I do have about this bug is, run the XT8 with beta3 and check for crashes..
So far so good.. no crashes on my side.. 
Should I do anything to get them?

hyprsnpr, 2023-01-15:
> @rlfrank165 Asus listen to the feedback sent to them and along with the feedback they get reports on crashes. They are really active in fixing stuff, but sometimes it's not easy, requires time and there is no clear info of what's the problem. As developer this report for instance doesn't tell me how to reproduce the issue, or at least for the time being I am not able to know what's the problem on your side.
> 
> Gents, please when you have issues please try to describe all the steps useful for me to test the problem itself and please attach logs. Moreover do not go off-topic. otherwise I have to read tons of comments and I do get lost.. As for now the only info I have about this bug is, to run the XT8 with beta3 and check for crashes, and so far so good. no crashes on my side, hence I am not able to reproduce this.
> 
> If you are vage I will keep on postponing the stable..
> 
> Can somebody tell me how to reproduce this? or can I get more details? Is the latest beta3 stable enough?

I wish I had more detail to provide, I didn't do anything special to make it happen. All I did is update to beta 3 and I started getting random crashes. Once I added swap the crashes stopped.

rlfrank165, 2023-01-15:
I share the frustration of others (both developers and users) who can't
provide anything more than random patterns of crashes and hangs with
nothing more specific in terms of what precipitated them.  At least at the
moment these are not noticeably caused by a singular event or pattern that
is visible.

All I know is that I first noticed these issues when I firmware upgraded to
the first 388 release officially made available by ASUS, which caused me to
switch to the 386 Merlin/Gnuton release for the XT8 (which was incredibly
reliable), and then problems re-appeared when I started using the (beta1)
388 Merlin/Gnuton release.  I haven't noticed much difference in terms of
these instabilities as I switched to beta2 and then beta3.  In my case they
typically occur after about 24 hours of uptime, but not on a predictable
time of day pattern.  But I have also had periods of 2 to 3 days where
everything is fine.

I have been doing factory resets (mostly thru the web UI but last time
actually a full hardware reset) on the router, and hope to get some time
today to do at least a UI based factory reset on the router and my 2
(ethernet connected) mesh nodes.  These problems seem to be reported by
people with both ethernet and wifi connected mesh nodes, so that doesn't
seem to be a factor.

My experience is that even the crashes/hangs are very different, sometimes
outright crashes, othertimes simply WiFi disconnects that spread and which
force me to reboot before an actual crash occurs.  I do export syslogs to a
desktop machine running a syslog server, and after every crash or needed
manual reboot do look at the logs to best that I am able, and never see
anything terribly unusual or suspect recorded.  I will start simply sending
to this list all logs so that maybe someone more sophisticated might see
something.  I have not turned on swap space, but am looking into trying
that.  I assume that turning on swap space solving the problems might be
indicative of a memory leak that is the underlying cause, but even then
where the memory leak is occurring is still a mystery.

One log related item I have noticed, which I doubt is related to the
problem, is that often near a crash time I get a log entry like this (and
these occur periodically having nothing to do with crashes)
Daemon.InfoJan 15 13:17:35 Frank-A2Router-BB276DE-C hostapd: eth4: STA
2c:aa:8e:1e:63:08 RADIUS: starting accounting session 6D79C0707B80440E
Only reason I even mention this (and I am very familiar with what Radius is
having set up many Radius based environments in commercial environments I
have worked on) is that in my case I'm not using Radius (not using
enterprise WPA or anything that would require Radius) and don't to the best
that I can tell turn Radius on or set up a link to a Radius server), is to
ask if it's normal for Radius accounting sessions to even be present if
Radius isn't being used?  Just making sure that even this minimal Radius
log is normal if Radius isn't being used.  Seems a little out of the
ordinary.
Anyway, will try to continue to look a syslogs and forward anything that
might provide any help.  I really love everything else about the ASUS and
Merlin releases, as ASUS is one of the few home routers still being made
that gives me the kind of control over my network that I want.  Almost
every other vendor is moving so radically to dumbed down UI's and even
discontinuing web UIs totally in favor of totally dumbed down phone app
interfaces.  Won't even look at Netgear, Amazon, Google and Linksys
equipment any more.  ASUS clearly does the right thing by making systems
that are easy to use for those who want that, but still offering incredibly
fully functional ones for those who want expert level control over their
home environment.
Randy

gnuton, 2023-01-15:
@rlfrank165  I checked the conf and I was not able to find anything interesting there for radius for hostapd. 
hostapd comes with an integreated radius implementation and this is the reason why you see these messages

rlfrank165, 2023-01-17:
Most recent crash with hour before syslog; nothing I can see that looks at
all suspect other than these somewhat unusual "not mesh client" logs.  I
had noticed these did start appearing in some of the other syslogs around
the time of other crashes, so not sure if they are indicative of some
problem that builds and results in a crash.  Wasn't awake at the time so no
personal experience with what was happening with the home net when this
occurred.  I have the full syslogs for all hours before from before what
I'm including here if that would be helpful

Randy

Daemon.InfoJan 17 06:33:34 Frank-A2Router-BB276DE-C hostapd: eth4: STA
9c:8e:cd:13:dc:2c WPA: group key handshake completed (RSN)
User.WarningJan 17 06:45:46 Frank-A2Router-BB276DE-C kernel:
60:A4:B7:9F:0F:71 not mesh client, can't delete it
User.WarningJan 17 06:48:46 Frank-A2Router-BB276DE-C kernel:
00:5F:67:AF:0F:C8 not mesh client, can't delete it
User.WarningJan 17 07:05:23 Frank-A2Router-BB276DE-C kernel:
00:5F:67:AF:0F:C8 not mesh client, can't delete it
User.NoticeJan 17 07:07:28 Frank-A2Router-BB276DE-C wlceventd:
wlceventd_proc_event(469): eth4: Deauth_ind 2C:9F:FB:8A:53:44, status: 0,
reason: Unspecified reason (1)
User.NoticeJan 17 07:07:28 Frank-A2Router-BB276DE-C wlceventd:
wlceventd_proc_event(505): eth4: Auth 2C:9F:FB:8A:53:44, status: Successful
(0)
User.NoticeJan 17 07:07:28 Frank-A2Router-BB276DE-C wlceventd:
wlceventd_proc_event(534): eth4: Assoc 2C:9F:FB:8A:53:44, status:
Successful (0)
Daemon.InfoJan 17 07:07:28 Frank-A2Router-BB276DE-C hostapd: eth4: STA
2c:9f:fb:8a:53:44 IEEE 802.11: associated
Daemon.InfoJan 17 07:07:28 Frank-A2Router-BB276DE-C hostapd: eth4: STA
2c:9f:fb:8a:53:44 RADIUS: starting accounting session 1B5F5943337C8F26
Daemon.InfoJan 17 07:07:28 Frank-A2Router-BB276DE-C hostapd: eth4: STA
2c:9f:fb:8a:53:44 WPA: pairwise key handshake completed (RSN)
Daemon.InfoJan 17 07:07:33 Frank-A2Router-BB276DE-C dnsmasq-dhcp[14505]:
DHCPDISCOVER(br0) 2c:9f:fb:8a:53:44
Daemon.InfoJan 17 07:07:33 Frank-A2Router-BB276DE-C dnsmasq-dhcp[14505]:
DHCPOFFER(br0) 192.168.10.70 2c:9f:fb:8a:53:44
Daemon.InfoJan 17 07:07:33 Frank-A2Router-BB276DE-C dnsmasq-dhcp[14505]:
DHCPREQUEST(br0) 192.168.10.70 2c:9f:fb:8a:53:44
Daemon.InfoJan 17 07:07:33 Frank-A2Router-BB276DE-C dnsmasq-dhcp[14505]:
DHCPACK(br0) 192.168.10.70 2c:9f:fb:8a:53:44 Somfy_ff11
User.WarningJan 17 07:07:34 Frank-A2Router-BB276DE-C kernel:
2C:9F:FB:8A:53:44 not mesh client, can't update it's ip
Daemon.InfoJan 17 07:07:34 Frank-A2Router-BB276DE-C dnsmasq-dhcp[14505]:
DHCPDISCOVER(br0) 2c:9f:fb:8a:53:44
Daemon.InfoJan 17 07:07:34 Frank-A2Router-BB276DE-C dnsmasq-dhcp[14505]:
DHCPOFFER(br0) 192.168.10.70 2c:9f:fb:8a:53:44
Daemon.InfoJan 17 07:07:34 Frank-A2Router-BB276DE-C dnsmasq-dhcp[14505]:
DHCPREQUEST(br0) 192.168.10.70 2c:9f:fb:8a:53:44
Daemon.InfoJan 17 07:07:34 Frank-A2Router-BB276DE-C dnsmasq-dhcp[14505]:
DHCPACK(br0) 192.168.10.70 2c:9f:fb:8a:53:44 Somfy_ff11
User.WarningJan 17 07:07:34 Frank-A2Router-BB276DE-C kernel:
2C:9F:FB:8A:53:44 not mesh client, can't update it's ip
Daemon.InfoJan 17 07:11:59 Frank-A2Router-BB276DE-C dnsmasq-dhcp[14505]:
DHCPDISCOVER(br0) 14:3f:a6:ed:17:0d
Daemon.InfoJan 17 07:11:59 Frank-A2Router-BB276DE-C dnsmasq-dhcp[14505]:
DHCPOFFER(br0) 192.168.10.96 14:3f:a6:ed:17:0d
Daemon.InfoJan 17 07:11:59 Frank-A2Router-BB276DE-C dnsmasq-dhcp[14505]:
DHCPREQUEST(br0) 192.168.10.96 14:3f:a6:ed:17:0d
Daemon.InfoJan 17 07:11:59 Frank-A2Router-BB276DE-C dnsmasq-dhcp[14505]:
DHCPACK(br0) 192.168.10.96 14:3f:a6:ed:17:0d
Daemon.InfoJan 17 07:15:58 Frank-A2Router-BB276DE-C dnsmasq-dhcp[14505]:
DHCPREQUEST(br0) 192.168.10.245 d8:f1:5b:b4:6d:0b
Daemon.InfoJan 17 07:15:58 Frank-A2Router-BB276DE-C dnsmasq-dhcp[14505]:
DHCPACK(br0) 192.168.10.245 d8:f1:5b:b4:6d:0b ESP_B46D0B
Daemon.InfoJan 17 07:15:59 Frank-A2Router-BB276DE-C dnsmasq-dhcp[14505]:
DHCPREQUEST(br0) 192.168.10.246 d8:f1:5b:e1:b2:ef
Daemon.InfoJan 17 07:15:59 Frank-A2Router-BB276DE-C dnsmasq-dhcp[14505]:
DHCPACK(br0) 192.168.10.246 d8:f1:5b:e1:b2:ef ESP_E1B2EF
[... 157 lines cut ...]
192.168.10.138.
User.NoticeJan 17 08:01:23 Frank-A2Router-BB276DE-C wlceventd:
wlceventd_proc_event(534): wl0.1: Assoc A4:DA:22:2C:09:5D, status:
Successful (0)

crash

Daemon.InfoMay  5 01:05:29 Frank-A2Router-BB276DE-C avahi-daemon[1824]:
Joining mDNS multicast group on interface br0.IPv4 with address
192.168.10.1.
Daemon.InfoMay  5 01:05:29 Frank-A2Router-BB276DE-C avahi-daemon[1824]: New
relevant interface br0.IPv4 for mDNS.
Daemon.InfoMay  5 01:05:29 Frank-A2Router-BB276DE-C avahi-daemon[1824]:
Joining mDNS multicast group on interface lo.IPv4 with address 127.0.1.1.
Daemon.InfoMay  5 01:05:29 Frank-A2Router-BB276DE-C avahi-daemon[1824]: New
relevant interface lo.IPv4 for mDNS.
Daemon.InfoMay  5 01:05:29 Frank-A2Router-BB276DE-C avahi-daemon[1824]:
Network interface enumeration completed.
Daemon.InfoMay  5 01:05:29 Frank-A2Router-BB276DE-C avahi-daemon[1824]:
Registering new address record for 192.168.10.1 on br0.IPv4.
Daemon.InfoMay  5 01:05:29 Frank-A2Router-BB276DE-C avahi-daemon[1824]:
Registering new address record for 127.0.1.1 on lo.IPv4.

Message ID: ***@***.***>
>

gnuton, 2023-01-17:
what's a crash? is the restart of the router?

rlfrank165, 2023-01-18:
I put that in there: it's the break in the syslog when the router
re-started itself.  I didn't manually reboot it but the kernel I assumed
decided it needed to re-boot.

The fact that there is nothing of real interest in the syslog before it is
why I am suspect that some syslog entries aren't making it to the remote
IP-based syslog server, which is why I was asking before if there is some
alternative way to capture syslogs on the router itself in NVRAM from just
before a crash that might be getting lost as they aren't being sent to a
remote IP-based syslog server since the router kernel is obviously not in
very good shape at that point.  I am admittedly reaching the edge of my
expertise in this matter.  Sorry.  I'm willing to try and do what people a
lot more expert than me would like to be able to help out in trying to sort
this out.

Randy

Randy Frank

rlfrank165, 2023-01-18:
What's weird is the syslog msg timing

last msg logged on remote syslog server from before crash/reboot
User.NoticeJan 17 08:01:23 Frank-A2Router-BB276DE-C wlceventd:
wlceventd_proc_event(534): wl0.1: Assoc A4:DA:22:2C:09:5D, status:
Successful (0)

first msg logged upon new boot (date/time I assume is from some element of
kernel build, before NTP runs
Daemon.InfoMay  5 01:05:29 Frank-A2Router-BB276DE-C avahi-daemon[1824]:
Joining mDNS multicast group on interface br0.IPv4 with address
192.168.10.1.

last syslog msg before NTP syncs clock, so 30 seconds from reboot to NTP
running, not atypical
Daemon.InfoMay  5 01:05:59 Frank-A2Router-BB276DE-C hostapd: eth4: STA
00:5f:67:ae:ff:00 WPA: pairwise key handshake completed (RSN)
Daemon.InfoJan 17 08:13:39 Frank-A2Router-BB276DE-C hostapd: eth4: STA
60:a4:b7:9f:0f:5d RADIUS: starting accounting session 0A51995A17BC5A31

working backwards kernel rebooted/first syslog about 08:13:09 so no syslogs
on remote server from 08:01:23 to 08:13:09 or a little under 12 minutes.
Seems like a long time for no syslog entries even taking into account
typical reboot overhead.  As I said, this analysis is from general OS
knowledge, not a Linux kernel expert by any stretch of the imagination.

Randy

gnuton, 2023-01-18:
@rlfrank165 no problem. I was just trying to understand what a "crash" meant for you. 
At moment I do not see a crash (no stacktrace) nor a reboot (no boot logs).

The logs are not complete, there must be something in the middle which seems to be missing to me and can let us understand what's the problem.
another thing you can do when this happens is  to check the uptime of your router, whitch is something you can find here http://router.asus.com/Tools_Sysinfo.asp.

felalex, 2023-01-18:
> @rlfrank165 no problem. I was just trying to understand what a "crash" meant for you. At moment I do not see a crash (no stacktrace) nor a reboot (no boot logs).
>
> The logs are not complete, there [...]

Is there an option for extended telemetry and debug data to be saved locally (say on a thumbstick you can connect to the usb port) for beta firmwares? So that postmortem analysis can be performed with a better insight…

I think it’s more reliable then sending logs over the network, which relies on the router being able to do it during a crash.

gnuton, 2023-01-18:
@felalex the logs are already stored in a persistent partition which can be consulted after a crash.

rlfrank165, 2023-01-18:
I absolutely do check uptime; that's how I know that a reboot occurred:

[image: image.png]

So this says that a reboot occurred yesterday 1/17/23 at about 08:12:29
which is totally consistent with the syslogs I sent.  That's why my theory
is that something is degrading before the router actually reboots that is
preventing syslog entries from being sent to the remote syslog server.,
which requires that it be able to send the data over a TCP/IP connection to
the syslog server.  This is consistent with my actual experience when I am
using the net during one of these "incidents" which is that wifi devices
start disconnecting and dropping offline but that some remain operational,
and then after some period of time the router eventually reboots.  I know
this sounds a little crazy, but this appears to be a common experience with
these crashes.

That is why I was asking if there is some way of capturing syslog entries
on the router itself (probably in NVRAM) that isn't dependent on the router
being able to transmit syslog entries over the net to the remote syslog
server. so that after reboot one gets a full picture of what was happening
just prior to a crash/reboot.

Clearly whatever is going on here is complicated and difficult to diagnose,
witness all of the complaints on the ASUS community bulletin boards about
the instabilities in the 388 releases.

I'm trying trying to provide whatever I can to help get to the bottom of
this.

Randy

Nehaxak, 2023-01-20:
Hi, just thought I'd add this in from my own syslog, using same beta 3 on a Asus TUF ax5400 and noticed two crashes last few days. Only managed this time earlier tonight to pick something up from the log, not sure if it's of any great use but passing it on anyway if it's of any help. Will update to latest stable that was released earlier tonight, maybe that might fix whatever it is, thanks.

Syslog details...

...
Start
... 

Jan 19 23:28:52 kernel: Division by zero in kernel.
Jan 19 23:28:52 kernel: CPU: 1 PID: 655 Comm: wl0-kthrd Tainted: P           O    4.1.52 #1
Jan 19 23:28:52 kernel: Hardware name: Generic DT based system
Jan 19 23:28:52 kernel: [<c0026f20>] (unwind_backtrace) from [<c0022cf8>] (show_stack+0x10/0x14)
Jan 19 23:28:52 kernel: [<c0022cf8>] (show_stack) from [<c045a80c>] (dump_stack+0x8c/0xa0)
Jan 19 23:28:52 kernel: [<c045a80c>] (dump_stack) from [<c0203a18>] (Ldiv0+0x8/0x10)
Jan 19 23:28:52 kernel: [<c0203a18>] (Ldiv0) from [<bf419374>] (wlc_ampdu_taf_release+0x170/0x1ccc [wl])
Jan 19 23:28:52 kernel: [<bf419374>] (wlc_ampdu_taf_release [wl]) from [<bf502430>] (taf_ias_sched_scb+0x1344/0x1b68 [wl])
Jan 19 23:28:52 kernel: [<bf502430>] (taf_ias_sched_scb [wl]) from [<bf503498>] (taf_ias_schedule+0x554/0x1338 [wl])
Jan 19 23:28:52 kernel: [<bf503498>] (taf_ias_schedule [wl]) from [<bf4feed8>] (wlc_taf_schedule+0x13c/0x26c [wl])
Jan 19 23:28:52 kernel: [<bf4feed8>] (wlc_taf_schedule [wl]) from [<bf41077c>] (wlc_ampdu_txeval+0x118/0x1730 [wl])
Jan 19 23:28:52 kernel: [<bf41077c>] (wlc_ampdu_txeval [wl]) from [<bf418078>] (wlc_ampdu_agg_pktc+0x8c0/0xc1c [wl])
Jan 19 23:28:52 kernel: [<bf418078>] (wlc_ampdu_agg_pktc [wl]) from [<bf4b6c58>] (wlc_nar_transmit_packet+0xf0/0x2b8 [wl])
Jan 19 23:28:52 kernel: [<bf4b6c58>] (wlc_nar_transmit_packet [wl]) from [<bf51d3a4>] (wlc_sendpkt+0x998/0xc40 [wl])
Jan 19 23:28:52 kernel: [<bf51d3a4>] (wlc_sendpkt [wl]) from [<bf3e3a78>] (wl_pktc_tx+0x48/0x5c [wl])
Jan 19 23:28:52 kernel: [<bf3e3a78>] (wl_pktc_tx [wl]) from [<bf3e3ae4>] (wl_start_int_try.constprop.10+0x58/0xdc [wl])
Jan 19 23:28:52 kernel: [<bf3e3ae4>] (wl_start_int_try.constprop.10 [wl]) from [<bf3e53a8>] (wl_start_txqwork+0x78/0x128 [wl])
Jan 19 23:28:52 kernel: [<bf3e53a8>] (wl_start_txqwork [wl]) from [<bf273394>] (wl_worker_thread_func+0xb4/0x184 [wl])
Jan 19 23:28:52 kernel: [<bf273394>] (wl_worker_thread_func [wl]) from [<c00471fc>] (kthread+0xdc/0xf4)
Jan 19 23:28:52 kernel: [<c00471fc>] (kthread) from [<c001f5e8>] (ret_from_fork+0x14/0x2c)
Jan 19 23:28:52 kernel: Division by zero in kernel.
Jan 19 23:28:52 kernel: CPU: 1 PID: 14947 Comm: kworker/1:0 Tainted: P           O    4.1.52 #1
Jan 19 23:28:52 kernel: Hardware name: Generic DT based system
Jan 19 23:28:52 kernel: Workqueue: events wl_timer_task [wl]
Jan 19 23:28:52 kernel: [<c0026f20>] (unwind_backtrace) from [<c0022cf8>] (show_stack+0x10/0x14)
Jan 19 23:28:52 kernel: [<c0022cf8>] (show_stack) from [<c045a80c>] (dump_stack+0x8c/0xa0)
Jan 19 23:28:52 kernel: [<c045a80c>] (dump_stack) from [<c0203a18>] (Ldiv0+0x8/0x10)
Jan 19 23:28:52 kernel: [<c0203a18>] (Ldiv0) from [<bf419374>] (wlc_ampdu_taf_release+0x170/0x1ccc [wl])
Jan 19 23:28:52 kernel: [<bf419374>] (wlc_ampdu_taf_release [wl]) from [<bf502430>] (taf_ias_sched_scb+0x1344/0x1b68 [wl])
Jan 19 23:28:52 kernel: [<bf502430>] (taf_ias_sched_scb [wl]) from [<bf503498>] (taf_ias_schedule+0x554/0x1338 [wl])
Jan 19 23:28:52 kernel: [<bf503498>] (taf_ias_schedule [wl]) from [<bf4feed8>] (wlc_taf_schedule+0x13c/0x26c [wl])
Jan 19 23:28:52 kernel: [<bf4feed8>] (wlc_taf_schedule [wl]) from [<bf4ff3f0>] (taf_watchdog+0xf0/0x120 [wl])
Jan 19 23:28:52 kernel: [<bf4ff3f0>] (taf_watchdog [wl]) from [<bf3efaf8>] (wlc_watchdog+0x518/0x960 [wl])
Jan 19 23:28:52 kernel: [<bf3efaf8>] (wlc_watchdog [wl]) from [<bf3eff4c>] (wlc_watchdog_timer+0xc/0x50 [wl])
Jan 19 23:28:52 kernel: [<bf3eff4c>] (wlc_watchdog_timer [wl]) from [<bf3e1cac>] (_wl_timer+0xbc/0x128 [wl])
Jan 19 23:28:52 kernel: [<bf3e1cac>] (_wl_timer [wl]) from [<bf3e1d74>] (wl_timer_task+0x14/0x4c [wl])
Jan 19 23:28:52 kernel: [<bf3e1d74>] (wl_timer_task [wl]) from [<c00422e4>] (process_one_work+0x118/0x3e4)
Jan 19 23:28:52 kernel: [<c00422e4>] (process_one_work) from [<c00425e4>] (worker_thread+0x34/0x540)
Jan 19 23:28:52 kernel: [<c00425e4>] (worker_thread) from [<c00471fc>] (kthread+0xdc/0xf4)
Jan 19 23:28:52 kernel: [<c00471fc>] (kthread) from [<c001f5e8>] (ret_from_fork+0x14/0x2c)

... 
End
...

rlfrank165, 2023-01-20:
> Jan 19 23:28:52 kernel: [] (unwind_backtrace) from []
> Jan 19 23:28:52 kernel: [] (show_stack) from [] (dump_stack+0x8c/0xa0)
> Jan 19 23:28:52 kernel: [] (dump_stack) from [] (Ldiv0+0x8/0x10)
> Jan 19 23:28:52 kernel: [] (Ldiv0) from []
> Jan 19 23:28:52 kernel: [] (wlc_ampdu_taf_release [wl]) from []
> Jan 19 23:28:52 kernel: [] (taf_ias_sched_scb [wl]) from []
> Jan 19 23:28:52 kernel: [] (taf_ias_schedule [wl]) from []
> Jan 19 23:28:52 kernel: [] (wlc_taf_schedule [wl]) from []
> Jan 19 23:28:52 kernel: [] (wlc_ampdu_txeval [wl]) from []
> Jan 19 23:28:52 kernel: [] (wlc_ampdu_agg_pktc [wl]) from []
> Jan 19 23:28:52 kernel: [] (wlc_nar_transmit_packet [wl]) from []
> Jan 19 23:28:52 kernel: [] (wlc_sendpkt [wl]) from []
> Jan 19 23:28:52 kernel: [] (wl_pktc_tx [wl]) from []
> Jan 19 23:28:52 kernel: [] (wl_start_int_try.constprop.10 [wl]) from []
> Jan 19 23:28:52 kernel: [] (wl_start_txqwork [wl]) from []
> Jan 19 23:28:52 kernel: [] (wl_worker_thread_func [wl]) from []
> Jan 19 23:28:52 kernel: [] (kthread) from [] (ret_from_fork+0x14/0x2c)
> Jan 19 23:28:52 kernel: [] (taf_watchdog [wl]) from []
> Jan 19 23:28:52 kernel: [] (wlc_watchdog [wl]) from []
> Jan 19 23:28:52 kernel: [] (wlc_watchdog_timer [wl]) from []
> Jan 19 23:28:52 kernel: [] (_wl_timer [wl]) from []
> Jan 19 23:28:52 kernel: [] (wl_timer_task [wl]) from []
> Jan 19 23:28:52 kernel: [] (process_one_work) from []
> Jan 19 23:28:52 kernel: [] (worker_thread) from [] (kthread+0xdc/0xf4)

My recollection is that a few weeks ago someone else posted another log
pre-crash log that had a similar "division by zero in kernel" so maybe
there is something in common here that may explain these crashes.  For
those with Linux kernel expertise, what does the Linux kernel do in these
cases?  Is there any recovery attempt, or simply a reboot?

I'm upgrading to the stable release tonight along with reset to factory
defaults, so we'll see if anything stabilizes.

Randy

micron888, 2023-01-20:
388.1.0 stable

Jan 20 15:21:34 kernel: xhci-hcd xhci-hcd.0: Timeout while waiting for setup device command
Jan 20 15:21:50 kernel: xhci-hcd xhci-hcd.0: Timeout while waiting for setup device command
Jan 20 15:21:50 kernel: usb 2-1: device not accepting address 2, error -62
Jan 20 15:22:10 kernel: xhci-hcd xhci-hcd.0: Timeout while waiting for stop endpoint command
Jan 20 15:22:30 kernel: xhci-hcd xhci-hcd.0: Timeout waiting for reset device command
Jan 20 15:22:35 kernel: xhci-hcd xhci-hcd.0: ERROR Transfer event for disabled endpoint or incorrect stream ring
Jan 20 15:22:35 kernel: xhci-hcd xhci-hcd.0: @0000000012b3b950 00000000 00000000 1b000000 01058000
Jan 20 15:22:35 kernel: xhci-hcd xhci-hcd.0: Command completion event does not match command

gnuton, 2023-01-20:
@Nehaxak thanks a lot for this logs. Very useful.
The errors are in some kernel workers kworker (very generic) and wl0-kthrd
The errors defintely seems to come happens when sending some packets to the wireless interface wl0

that functions are in release/src-rt-5.02axhnd.675x/router-sysdep.tuf-ax5400/hnd_extra/prebuilt/wl.o
which are closed source. 

I will investigate more on this, but I am not sure I can do so much since that's a prebuilt and comes from ASUS.
In any case please update to the stable and reset to factory your routers.
If you see still this problem then let's see what to do.
Most of the times these are errors which comes from invalid configurations which you get when you flash firmware which use settings from previous versions.

micron888, 2023-01-20:
factory reset. format jffs
setup from scratch. survived overnight without restart.
will update again if run into issuees

rlfrank165, 2023-01-21:
Again, providing this just in case it might be useful to people looking
into the instability in 388.

Running stable Merlin/Gnuton 388, it's be up for and seems pretty stable.
From a user point of view the router seems fine, all clients seem to be
working just fine

[image: image.png]

But when I just logged into the router, saw it has zero clients (normally
have about +- 90

[image: image.png][image: 08acd0df-6046-4ba8-842d-8baa661361a4.png]

looked at the system and saw this (repeated as nasium)
[image: 5cf787e6-00a4-4caa-9416-c1e20ffcc65d.png]

repeated pretty constantly for the last 24+ hours, with no seeming effect
on the router stability.

Anyone have any idea what this means or might signify?

I'm attaching the full log in case anyone wants to take a deeper dive...
hoping that someone all of this may help to explain the
irreproducable crashes.

Randy

Jan 21 11:31:13 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 8 s)
Jan 21 11:31:14 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 6 s)
Jan 21 11:31:20 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 8 s)
Jan 21 11:31:29 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 8 s)
Jan 21 11:31:37 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:31:44 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:31:51 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 8 s)
Jan 21 11:31:59 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:32:06 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:32:13 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:32:16 dnsmasq-dhcp[4952]: DHCPREQUEST(br0) 192.168.10.238 b4:b0:24:ea:88:d7 
Jan 21 11:32:16 dnsmasq-dhcp[4952]: DHCPACK(br0) 192.168.10.238 b4:b0:24:ea:88:d7 KASAElectrical
Jan 21 11:32:16 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 5 s)
Jan 21 11:32:21 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 8 s)
Jan 21 11:32:29 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:32:32 dnsmasq-dhcp[4952]: DHCPREQUEST(br0) 192.168.10.223 00:31:92:3c:17:df 
Jan 21 11:32:32 dnsmasq-dhcp[4952]: DHCPACK(br0) 192.168.10.223 00:31:92:3c:17:df HS210
Jan 21 11:32:32 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 4 s)
Jan 21 11:32:36 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 8 s)
Jan 21 11:32:45 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:32:52 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:32:59 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:33:06 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:33:13 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 9 s)
Jan 21 11:33:22 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:33:29 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:33:36 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:33:43 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:33:51 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 8 s)
Jan 21 11:33:59 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:34:06 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:34:13 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 8 s)
Jan 21 11:34:21 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 9 s)
Jan 21 11:34:30 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:34:37 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 8 s)
Jan 21 11:34:44 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 2 s)
Jan 21 11:34:46 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:34:53 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 11:35:00 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 8 s)
[... cut ...]
Jan 21 12:26:39 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 8 s)
Jan 21 12:26:47 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 12:26:54 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 8 s)
Jan 21 12:27:02 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 7 s)
Jan 21 12:27:09 dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No space

rlfrank165, 2023-01-21:
Adding some add'l info.  This is the syslog entry (today at 00:19:58) when
this first appeared; interesting observation is that this began
occurring almost exactly after the router had been up 24 hours. This is the
first time I have ever seen this "no space left on device" and have no idea
what it means (what device?).  Was wrong it previous msg: problem had only
started about 16 hours ago at (00:09:58 1/21/23) after router had been up
24 hours, but has been repeating for last 16 hours.  No effect on perceived
router performance didn't even notice until I logged into router and noted
"0 clients"

Daemon.InfoJan 21 00:00:13 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: DHCPREQUEST(br0) 192.168.10.219 00:31:92:48:c5:42
Daemon.InfoJan 21 00:00:13 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: DHCPACK(br0) 192.168.10.219 00:31:92:48:c5:42
HS200
Daemon.InfoJan 21 00:01:07 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: DHCPREQUEST(br0) 192.168.10.74 04:21:4c:ad:e1:01
Daemon.InfoJan 21 00:01:07 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: DHCPACK(br0) 192.168.10.74 04:21:4c:ad:e1:01
energybridge2-1f38c9d4e31726bd
Daemon.InfoJan 21 00:03:14 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: DHCPREQUEST(br0) 192.168.10.205 b0:a7:b9:37:e1:8c
Daemon.InfoJan 21 00:03:14 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: DHCPACK(br0) 192.168.10.205 b0:a7:b9:37:e1:8c
HS220
Daemon.InfoJan 21 00:03:39 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: DHCPREQUEST(br0) 192.168.10.234 14:eb:b6:df:fc:fc
Daemon.InfoJan 21 00:03:39 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: DHCPACK(br0) 192.168.10.234 14:eb:b6:df:fc:fc EP40
Daemon.InfoJan 21 00:04:03 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: DHCPREQUEST(br0) 192.168.10.33 7c:78:b2:99:a4:3d
Daemon.InfoJan 21 00:04:03 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: DHCPACK(br0) 192.168.10.33 7c:78:b2:99:a4:3d
WyzeCam
Daemon.InfoJan 21 00:04:45 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: DHCPREQUEST(br0) 192.168.10.80 00:18:61:3d:ee:6b
Daemon.InfoJan 21 00:04:45 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: DHCPACK(br0) 192.168.10.80 00:18:61:3d:ee:6b
myx_0018613DEE6A
Daemon.InfoJan 21 00:06:53 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: DHCPREQUEST(br0) 192.168.10.95 04:a2:22:16:6d:62
Daemon.InfoJan 21 00:06:53 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: DHCPACK(br0) 192.168.10.95 04:a2:22:16:6d:62
Daemon.InfoJan 21 00:07:00 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: DHCPREQUEST(br0) 192.168.10.91 c8:3a:6b:21:15:21
Daemon.InfoJan 21 00:07:00 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: DHCPACK(br0) 192.168.10.91 c8:3a:6b:21:15:21
RokuUltra
User.ErrorJan 21 00:19:54 Frank-A2Router-BB276DE-C rc_service: dhcp6c
12172:notify_rc restart_ddns
User.ErrorJan 21 00:19:54 Frank-A2Router-BB276DE-C ddns: update
WWW.DYNDNS.ORG ***@***.***, wan_unit 0
User.ErrorJan 21 00:19:54 Frank-A2Router-BB276DE-C ddns: Clear ddns cache.
User.ErrorJan 21 00:19:54 Frank-A2Router-BB276DE-C ddns: Start Inadyn(10).
User.NoticeJan 21 00:19:54 Frank-A2Router-BB276DE-C inadyn[12180]:
In-a-dyn version 2.9.1 -- Dynamic DNS update client.
User.NoticeJan 21 00:19:55 Frank-A2Router-BB276DE-C inadyn[12180]:
Update forced for alias rlfrank.dyndns.org, new IP# 68.42.77.235
User.NoticeJan 21 00:19:55 Frank-A2Router-BB276DE-C inadyn[12180]:
Updating cache for rlfrank.dyndns.org
Daemon.InfoJan 21 00:19:58 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: DHCPREQUEST(br0) 192.168.10.200 60:a4:b7:9f:12:83
Daemon.InfoJan 21 00:19:58 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: DHCPACK(br0) 192.168.10.200 60:a4:b7:9f:12:83
HS220
Daemon.ErrorJan 21 00:19:58 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No
space left on device (retry in 5 s)
Daemon.ErrorJan 21 00:20:03 Frank-A2Router-BB276DE-C
dnsmasq-dhcp[4952]: failed to write /var/lib/misc/dnsmasq.leases: No
[... 36 lines cut ...]
Daemon.InfoJan 21 00:22:51 Frank-A2Router-BB276DE-C hostapd: eth5: STA
1c:fe:2b:94:82:2d WPA: group key handshake completed (RSN)
Daemon.InfoJan 21 00:22:51 Frank-A2Router-BB276DE-C hostapd: eth5: STA
50:dc:e7:a1:d6:ac WPA: group key handshake completed (RSN)
Daemon.InfoJan 21 00:22:51 Frank-A2Router-BB276DE-C hostapd: eth5: STA
a0:02:dc:e9:07:45 WPA: group key handshake completed (RSN)
Daemon.InfoJan 21 00:22:51 Frank-A2Router-BB276DE-C hostapd: eth4: STA
60:a4:b7:9f:4e:06 WPA: group key handshake completed (RSN)
Daemon.InfoJan 21 00:22:51 Frank-A2Router-BB276DE-C hostapd: eth4: STA
00:5f:67:af:0f:c8 WPA: group key handshake completed (RSN)
Daemon.InfoJan 21 00:22:51 Frank-A2Router-BB276DE-C hostapd: eth4: STA
2c:aa:8e:1e:63:08 WPA: group key handshake completed (RSN)
Daemon.InfoJan 21 00:22:51 Frank-A2Router-BB276DE-C hostapd: eth4: STA
b4:b0:24:ea:88:d7 WPA: group key handshake completed (RSN)
Daemon.InfoJan 21 00:22:51 Frank-A2Router-BB276DE-C hostapd: eth4: STA
60:a4:b7:82:59:97 WPA: group key handshake completed (RSN)
Daemon.InfoJan 21 00:22:51 Frank-A2Router-BB276DE-C hostapd: eth4: STA
00:31:92:d9:66:39 WPA: group key handshake completed (RSN)
Daemon.InfoJan 21 00:22:51 Frank-A2Router-BB276DE-C hostapd: eth4: STA
00:31:92:48:c8:7e WPA: group key handshake completed (RSN)
Daemon.InfoJan 21 00:22:51 Frank-A2Router-BB276DE-C hostapd: eth4: STA
00:31:92:d9:bd:f8 WPA: group key handshake completed (RSN)

Daemon.InfoJan 21 00:22:51 Frank-A2Router-BB276DE-C hostapd: eth4: STA
00:5f:67:af:00:c0 WPA: group key handshake completed (RSN)

gnuton, 2023-01-23:
@rlfrank165
0 clients usually happens when your browser cache the web page but the api fails because the auth token is expired.
Basically logout and login back to get it working.

 It seems you have something which is consuming all your disk space in /var
 you can try to run 
 find . -type f -size +1M to see what are the fatty files