gnuton/asuswrt-merlin.ng #322: Remote access via WAN

dacootmeister, 2023-01-13:
Router Model Affected
Models: RT-AX95Q

Firmware Version Affected
388.1-gnuton0_beta3

Is this bug present in upstream Merlin releases too?
Not sure.

Describe the bug
When **enabling** the option in System, Remote Access Config, _Enable Web Access from WAN_ the router (first giving a warning), saves the config, but it doesn't get enabled. Refreshing the webpage renders the option to be disabled again.

To Reproduce
Steps to reproduce the behavior:
Follow the bug descripton

Expected behavior
I would expect this option to be available/saveable, just like in previous beta's

gnuton, 2023-01-13:
do you have any script installed in /jffs? is factory resetting the router fixing this?

dacootmeister, 2023-01-13:
@gnuton yes I have several installed in /jffs/scripts

YazDHCP              openvpn-event        services-stop
connmon              post-mount           spdmerlin
dnsmasq.postconf     pre-mount            uiDivStats
firewall-start       scmerlin             unmount
init-start           service-event        update-notification
nat-start            service-event-end
ntpmerlin            services-start

Resetting fixed the issue. 
No bug after all , sorry!

gnuton, 2023-01-13:
awesome! :D no problem closing

closed 2023-01-13 by gnuton