gnuton/asuswrt-merlin.ng #323: GameFirst V6 doesn't detect Asus TUF-AX5400 in the ''Smart Router'' section.

4Liberty, 2023-01-13:
**Router Model Affected**
Models: Asus TUF-AX5400

**Firmware Version Affected**
388.1_0-gnuton0_beta3

**Describe the bug**
GameFirst application doesn't detect Asus TUF-AX5400 router in the ''Smart Router'' section. It says ''No ASUS ROG FIRST supported routers can be found within range.'' However, this feature was working on the stable 386.07_2-gnuton1 firmware.

**To Reproduce**
Steps to reproduce the behavior:
1. Download and open the GameFirst V6.1.16.2 application.
2. Go to the Smart Router feature and click on the ''Discover'' button.
3. See the error text.

**Screenshots**
[image]

gnuton, 2023-01-14:
this is a TUG and nota ROG device.
Was the stock firmware being detected?

4Liberty, 2023-01-15:
> this is a TUF and nota ROG device. Was the stock firmware being detected?

Unfortunately, I didn't try it on the stock firmware but your stable 386.07_2-gnuton1 firmware was detecting it.

gnuton, 2023-01-15:
which version of TUF image have you flashed? the one with TUF theme or the classic one?

4Liberty, 2023-01-18:
> which version of TUF image have you flashed? the one with TUF theme or the classic one?

Classic one.

gnuton, 2023-01-18:
can you test the one with TUF ui?

4Liberty, 2023-01-23:
> can you test the one with TUF ui?

Today I tested with TUF UI but it is still not detected. I will also reinstall GameFirst and try again.