gnuton/asuswrt-merlin.ng #325: no 5ghz Guest Wifi available

gittyuphaz, 2023-01-15:
**Router Model Affected**
Models: RT-AX92U
Firmware Version Affected**
388.xx Includes 388.1-gnuton0_beta2

**Is this bug present in upstream Merlin releases too?**
Unsure. I updated to wrt merlin after install

**Describe the bug**
Unable to see 5ghz networks on guest network page. Only 2.5ghz and Facebook WiFi Appear.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Guest Network
2. See no 5ghz

**Expected behavior**
Scross down you should be able to dd guest got 5ghz

**Screenshots**

[image]

gnuton, 2023-01-15:
Hi,
have you reset to factory after updating from stock?

gittyuphaz, 2023-01-16:
I just tried that and can see the 5ghz guest option after unfortunately, if I then restore my router settings backup file it disappeared again.

gnuton, 2023-01-16:
@gittyuphaz stock and merlin do not have compatible settings. If this is what you are trying to do you basically can't without running in issues like this.
Can i close this issue?

gittyuphaz, 2023-01-16:
Hi, makes sense. thanks for the great support. plesase go ahead and close. All the best and thanks for your hard work on this fork!

closed 2023-01-16 by gittyuphaz