gnuton/asuswrt-merlin.ng #329: Latest 388.1 has destroyed by router

Ichimikichiki, 2023-01-21:
**Router Model Affected**
Models: TUF-AX3000

**Firmware Version Affected**
388.1

**Is this bug present in upstream Merlin releases too?**
idk

**Describe the bug**
This is my 14th hard reset. The router just red lights every time I change a wireless setting. Like ap isolated, power level etc. Or even just random stuff. And then goes on a reboot frenzy until you hard reset. There's no way to even log in and get logs. Is there any way to roll back??

gnuton, 2023-01-21:
I am sorry to hear this. The firmware has been tested for a few months.
I am not sure what happened to you, if there is a bad flash or what I will try to verify again on my router to see if there is any real problem.
Could you please tell me which version u used? the one with or without tuf ui?
You can roll back using the Asus restoration tool https://www.asus.com/support/FAQ/1000814/
or using TFTP if you now how to use it.

if hard reset works and you can get back to web ui then then you can simply flash a different firmware from the web ui.

gnuton, 2023-01-21:
I tested the firmware (with tuf ui) on my tuf ax300 v1 again changed wireless settings. all good.

Ichimikichiki, 2023-01-24:
All good. After 44 hard resets, I decided to try rolling back to the previous firmware. Which accepted it without issue. Not sure what the go is.

Try changing regions to Australia. Or try setting ap isolate to yes. It starts to work and after a minute it starts kicking you off and eventually just gets to a point where you can't even connect.

gnuton, 2023-01-24:
did you went back to stock?

closed 2024-02-25 by gnuton