gnuton/asuswrt-merlin.ng #330: Latest 388.1 stable (non TUF Uí), Asus TUF ax5400, brief router hang, kernel RT Throttling

Nehaxak, 2023-01-23:
**Router Model Affected**
Asus TUF ax5400

**Firmware Version Affected**
388.1 (non TUF version of firmware) 

**Is this bug present in upstream Merlin releases too?**

Unknown

**Describe the bug**

The router seemed to hang and then recovered itself less than a minute later, all wireless AP's (1 X 2.4ghz and 1 x 5ghz) disconnected. WAN disconnected, LAN also. Just seemed to freeze. 

There was only one line in the logs, as follows... 

---
Start
---

Jan 22 22:23:19 kernel: sched: RT throttling activated

---
End
---

**To Reproduce**

Unknown how to reproduce. 
Factory reset was done after firmware install and jffs wipe etc., all the usual stuff. 
Router has otherwise been working fine with the latest 388.1 firmware (non TUF Ui) version. 

Reading around a bit I've noticed the issue (or similar) has appeared for others in the past on older asuswrt versions and something to do with STP (Spanning Tree Protocol) suggesting disabling it helped stop this from happening. So I just checked and noticed STP was enabled by default after factory reset so I'll disable that now and reboot (I've no need for it anyway, no switches/hubs other than the router itself). 

Will see how it goes. If it happens again I'll post up any logs etc but just passing/reporting this issue on for your info if it is of any help at all and thank you :) 

**Expected behavior**

N/A

**Screenshots**

N/A

gnuton, 2023-01-23:
Thanks for reporting this.
I do have not investigated on this yet, but all your findings seems alrady quite useful as starting point.
Next steps are to understand why the router is RT throtling. 
Is it becasue there is malicious activity in the network which generates some packets with loops that are blocked by STP, or is the STP/or-somerhing-else which doesn't like works correcly.

vladut123, 2023-01-23:
> Thanks for reporting this.
> I do have not investigated on this yet, but all your findings seems alrady quite useful as starting point.
> Next steps are to understand why the router is RT throtling.
> Is it becasue there is malicious activity in the network which generates some packets with loops that are blocked by STP, or is the STP/or-something-else which doesn't like works correcly.

Nehaxak, 2023-01-23:
Yeah don't worry about it much at all, just wanted to post it up for your info really or in case anyone else experienced similar.

There's not a lot of info to go on from the logs anyway, was just a single line posted and sparse info to go on from that too. 

I've disabled STP and router's been up 8+ hours with zero issues since.

You can close this issue if you want as there's not a lot of detailed info available, can always revert back to this anyway if it starts happening again regularly and if I can get better logs and ideas on what or why it's happening.

Thanks :)

gnuton, 2023-01-23:
@Nehaxak I do just feel weird the problem can be in a kernel component since linux kernel should be pretty much stable. 
If ther is a problem it usually comes from closed source drivers.. 
But in any case thanks for this good description and let's keep an eye if this issue comes back