gnuton/asuswrt-merlin.ng #335: WAN link down, but it isn't

Hello-olleH, 2023-01-28:
**Router Model Affected**
DSL-AX82U

**Firmware Version Affected**
388.1_0-gnuton1 

Wrong DSL down Info. DSL is not down, but router says WAN is down. After a reboot, it will be fine again.

Logs:
﻿Jan 28 18:20:30 kernel: [ 8572.825912] Line 0: Rate Change, us=34589, ds=96203
Jan 28 18:20:30 kernel: [ 8572.830983] bcmxtmcfg: ULS Port 0 
Jan 28 18:20:30 kernel: [ 8572.838988] bcmxtmcfg: XTM Link Mode is VDSL G.inpRtx 
Jan 28 18:20:30 kernel: [ 8572.857360] bcmxtmcfg:ulPhysPort[0] m_ulTrafficMonitorPort[0]
Jan 28 18:20:30 kernel: [ 8572.863702] bcmxtmcfg: XTM LinkInfo. P-0, S-UP(1), Service =PTM(2) 
Jan 28 18:20:30 kernel: [ 8572.871983] bcmxtmcfg: ReconfSAR P-0, T-2 
Jan 28 18:20:30 kernel: [ 8572.879449] bcmxtmcfg: ChkSILI Port = 0, BondingPort = 4 
Jan 28 18:20:30 kernel: [ 8572.888948] bcmxtmcfg: DSL link mode is 4
Jan 28 18:20:30 kernel: [ 8572.895530] DSL LinkMode 4
Jan 28 18:20:30 kernel: [ 8572.900557] message received before monitor task is initialized kerSysSendtoMonitorTask 
Jan 28 18:26:05 kernel: [ 8907.634444] Line 0: Rate Change, us=34589, ds=96172
Jan 28 18:26:05 kernel: [ 8907.640988] bcmxtmcfg: ULS Port 0 
Jan 28 18:26:05 kernel: [ 8907.645496] bcmxtmcfg: XTM Link Mode is VDSL G.inpRtx 
Jan 28 18:26:05 kernel: [ 8907.652558] bcmxtmcfg:ulPhysPort[0] m_ulTrafficMonitorPort[0]
Jan 28 18:26:05 kernel: [ 8907.659990] bcmxtmcfg: XTM LinkInfo. P-0, S-UP(1), Service =PTM(2) 
Jan 28 18:26:05 kernel: [ 8907.672208] bcmxtmcfg: ReconfSAR P-0, T-2 
Jan 28 18:26:05 kernel: [ 8907.677029] bcmxtmcfg: ChkSILI Port = 0, BondingPort = 4 
Jan 28 18:26:05 kernel: [ 8907.689014] bcmxtmcfg: DSL link mode is 4
Jan 28 18:26:05 kernel: [ 8907.696706] DSL LinkMode 4
Jan 28 18:26:05 kernel: [ 8907.706009] message received before monitor task is initialized kerSysSendtoMonitorTask 
Jan 28 18:29:48 WAN_Connection: WAN(0) link down.
Jan 28 18:29:50 WAN_Connection: WAN was restored.
Jan 28 18:29:50 dnsmasq[2903]: read /etc/hosts - 8 addresses
Jan 28 18:29:50 dnsmasq[2903]: using nameserver 8.8.8.8#53
Jan 28 18:29:50 dnsmasq[2903]: using nameserver 1.1.1.1#53
Jan 28 18:29:50 custom_script: Running /jffs/scripts/nat-start
Jan 28 18:30:22 WAN_Connection: WAN(0) link down.

[image]

gnuton, 2023-01-29:
Please reset to factory and do not restore any backup.
This really feels like something in your config is messing up your router.
The logs do not say anything interesting.

Hello-olleH, 2023-01-29:
After the 388.1_0-gnuton1 update, I immediately performed a factory reset.
I always do this after every update and do not restore a backup.

gnuton, 2023-01-30:
@Hello-olleH can you format  disable the /jffs scripts? 
from the logs I see you ahve  /jffs/scripts/nat-start

Hello-olleH, 2023-01-30:
After the factory settings I once formatted jffs and then I copied my script there.
I use nat-start for WireGuard port forwarding. (BitTorrent) :-) 

[image]

gnuton, 2023-01-31:
if you router can ping IP addresses but not domain servers (eg: something.com)
the problem is in the DNS configuration, that somehow doesn't work all the times for some reasons (router or provider)?

can you just remove the custom DNS or add more fallback DNS to your configs?

R3D-PLAN3T, 2023-03-02:
Mar  2 21:04:31 kernel: [ 7676.201259] 
Mar  2 21:04:31 kernel: [ 7676.232441] 
Mar  2 21:04:31 kernel: [ 7676.619541] 
Mar  2 21:04:31 kernel: [ 7676.700597] 
Mar  2 21:04:33 kernel: [ 7678.546298] 
Mar  2 21:04:34 kernel: [ 7679.088995] 
Mar  2 21:04:34 kernel: [ 7679.289858] 
Mar  2 21:17:53 kernel: [ 8477.967549] Line 0: Rate Change, us=34433, ds=94674
Mar  2 21:17:53 kernel: [ 8477.972464] bcmxtmcfg: ULS Port 0 
Mar  2 21:17:53 kernel: [ 8477.978639] bcmxtmcfg: XTM Link Mode is VDSL G.inpRtx 
Mar  2 21:17:53 kernel: [ 8477.985893] bcmxtmcfg:ulPhysPort[0] m_ulTrafficMonitorPort[0]
Mar  2 21:17:53 kernel: [ 8477.999902] bcmxtmcfg: XTM LinkInfo. P-0, S-UP(1), Service =PTM(2) 
Mar  2 21:17:53 kernel: [ 8478.010066] bcmxtmcfg: ReconfSAR P-0, T-2 
Mar  2 21:17:53 kernel: [ 8478.014517] bcmxtmcfg: ChkSILI Port = 0, BondingPort = 4 
Mar  2 21:17:53 kernel: [ 8478.024204] bcmxtmcfg: DSL link mode is 4
Mar  2 21:17:53 kernel: [ 8478.028809] DSL LinkMode 4
Mar  2 21:17:53 kernel: [ 8478.033473] message received before monitor task is initialized kerSysSendtoMonitorTask 
Mar  2 21:21:16 kernel: [ 8681.156170] 
Mar  2 21:21:16 kernel: [ 8681.277810] 
Mar  2 21:21:16 kernel: [ 8681.376468] 
Mar  2 21:21:16 kernel: [ 8681.705649] 
Mar  2 21:21:17 kernel: [ 8682.469994] 
Mar  2 21:21:17 kernel: [ 8682.550422] 
Mar  2 21:22:37 kernel: [ 8762.917873] Line 0: Rate Change, us=34433, ds=94643
Mar  2 21:22:37 kernel: [ 8762.923340] bcmxtmcfg: ULS Port 0 
Mar  2 21:22:37 kernel: [ 8762.926779] bcmxtmcfg: XTM Link Mode is VDSL G.inpRtx 
Mar  2 21:22:37 kernel: [ 8762.932686] bcmxtmcfg:ulPhysPort[0] m_ulTrafficMonitorPort[0]
Mar  2 21:22:37 kernel: [ 8762.943838] bcmxtmcfg: XTM LinkInfo. P-0, S-UP(1), Service =PTM(2) 
Mar  2 21:22:37 kernel: [ 8762.954243] bcmxtmcfg: ReconfSAR P-0, T-2 
Mar  2 21:22:38 kernel: [ 8762.961171] bcmxtmcfg: ChkSILI Port = 0, BondingPort = 4 
Mar  2 21:22:38 kernel: [ 8762.974136] bcmxtmcfg: DSL link mode is 4
Mar  2 21:22:38 kernel: [ 8762.978782] DSL LinkMode 4
Mar  2 21:22:38 kernel: [ 8762.984060] message received before monitor task is initialized kerSysSendtoMonitorTask 
Mar  2 21:26:03 WAN_Connection: Fail to connect with some issues.
Mar  2 21:26:08 WAN_Connection: WAN was restored.
Mar  2 21:26:08 dnsmasq[4108]: read /etc/hosts - 8 addresses
Mar  2 21:26:08 dnsmasq[4108]: using nameserver 8.8.8.8#53
Mar  2 21:26:08 dnsmasq[4108]: using nameserver 1.1.1.1#53
Mar  2 21:26:09 custom_script: Running /jffs/scripts/nat-start
Mar  2 21:26:43 WAN_Connection: Fail to connect with some issues.

R3D-PLAN3T, 2023-03-14:
@gnuton  I find that Problem!
I don't use the default "Stability Adjustment" and for that I set the value to "-6"
The problem does not appear with default. 
When I have noise in my connection and the connection is reestablished, the router can no longer understand whether it is connected to the Internet.

Although in 386.x I don't have the problem and can just use "-6" value.
With "-6" I get about 20Mbit more speed.

closed 2023-03-14 by R3D-PLAN3T