gnuton/asuswrt-merlin.ng #336: Cake on DSL-AX82U needs looked at

gameinn1, 2023-01-30:
**Router Model Affected**
Models: DSL-AX82U

**Firmware Version Affected**
I test on Stable: 388.1_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
I have no idea sorry

**Describe the bug**
A clear and concise description of what the bug is.

I remember in some changelogs that the DSL-AX82U had problems with Cake. Apparently it was fixed in 386.03_2-gnuton0_beta5 but I believe there are still issues somewhere.

**To Reproduce**
Steps to reproduce the behavior:
1. Ping a well known IP you know is stable. For simple sake we say Google DNS with -t command
2. Verify that pings are stable for 10 seconds
3. Navigate to fast.com which will initiate a speed test
4. See that the pings in the cmd have jumped up when the test was initiated. Sometimes 1 ping but can last for 2 pings and very rarely 3 pings. To help explain, here is a screenshot: https://i.imgur.com/EKY5vZS.png
5. As you can see on ping 12 when the test was started in Chrome, the ping jumped up by approximately 20ms.

**Expected behavior**
Ping 12 would stay relatively close to the 16ms average. Proof that cake should hold stable ping even at 1Gbps timestamped here: https://youtu.be/t5xuTy1xn64?t=622

The DSL-AX82U not being able to handle it at less than 45 Mbps seems like a bug.

**Screenshots**
See above step 4

gnuton, 2023-01-30:
Hi,
Why do you think a good QoS config would give more priority to an ICMP
request instead of a HTTP stream (TCP?)
What's the use case?

gameinn1, 2023-01-30:
Don't get me wrong what I'm stating is very peculiar but the fact is the YouTube video linked above shows that a proper implementation of Cake should not have that issue even when pumping out close to 1Gbps even when ICMP is pinging 8.8.8.8.

Now granted the technology is far different (this person is obviously on some form of FTTP) vs this FTTC comparison but it's still very strange.

If your own testing on other routers (not Asus) shows ping spikes on ICMP when speedtests are done maybe I'm asking too much of it.

gnuton, 2023-01-31:
@gameinn1 you can use some CLI tools like tc-cake and tweak the config of the queues if you wanna achieve a particular behaviour. 
Here some tools you may find interesting

https://man7.org/linux/man-pages/man8/tc-cake.8.html#PRIORITY_QUEUE_PARAMETERS
https://github.com/ttgapers/cakeqos-merlin

gameinn1, 2023-01-31:
Thank you. I think I'll just leave it as is. That stuff seems too complex for me and the integrated version of Cake did what I originally wanted. A+ results in dslreports and waveform so I can't be too far from perfect.

gnuton, 2023-01-31:
I will close this then.
The Cake config comes straight from upstream. if there is any change in merlin then it will reflected in my builds too.

closed 2023-01-31 by gnuton