gnuton/asuswrt-merlin.ng #337: Router freezes when iptables TEE is used (traffic/port mirroring)

hjotha, 2023-01-31:
**Router Model Affected**
RT-AX82U

**Firmware Version Affected**
388.1_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
Router freezes when traffic mirroring is enabled via iptables.

**To Reproduce**
1. ssh in to the router
2.To enable traffic mirroring to certain IPs, the following commands can be used:

modprobe xt_TEE

# for specific host
iptables -t mangle -A PREROUTING  -s 192.168.1.64 -j TEE --gateway 192.168.1.137
iptables -t mangle -A POSTROUTING -d 192.168.1.64 -j TEE --gateway 192.168.1.137

OR

# for all traffic
iptables -t mangle -A PREROUTING  -j TEE --gateway 192.168.1.137
iptables -t mangle -A POSTROUTING -j TEE --gateway 192.168.1.137

Router is freezing after these commands are issued.

**Expected behavior**
Mirror all traffic to the ip specified in --gateway <ip>

gnuton, 2023-01-31:
For freeze do you mean the kernel doesn't respond anymore?
or you just lose the SSH connection?

hjotha, 2023-01-31:
ssh connection drops, can't ssh to the router anymore nor access webui or any other services running in it.

gnuton, 2023-01-31:
is 192.168.1.64 your source address?

hjotha, 2023-01-31:
yes it was but I cannot connect with other ips to the router also after the commands are issued