gnuton/asuswrt-merlin.ng #338: RX-AX89X -

mcarbonneaux, 2023-02-05:
RX-AX89X - https://www.asus.com/Networking-IoT-Servers/WiFi-Routers/ASUS-Gaming-Routers/RT-AX89X/

Current Version: 3.0.0.4.386.45130

status: **WONT_BE_SUPPORTED**
reason: not a broadcom device

what is blocking the fact that is qualcomm in place of broadcom ?

because this router are the only one from asus to have 10g spf+ for wan/lan that permit to connect to networkoperator directly in fiber in 8/10g plan.

gnuton, 2023-02-06:
short answer: my time.
long answer: router firmwares are based on SDKs coming from vendors. The one we largely use in Merlin is from broadcom.
Supporting a qualcomm SDK needs time to invest in learning and fixing stuff when they break.  I think it may make sense to invest time if you have a lot of users and devices, but this is not the case. 
I am not even sure Asus would support us for this router ( I have not even asked)

mcarbonneaux, 2023-02-21:
ok is not technical raison. i understand.
but i'm sade ! because this router seem to be promissing on 10G router...

gnuton, 2023-02-21:
If you like you can ask asus for the GPL and you can try to merge the GPL into Merlin yourself.
This is the way I started with this project :D

closed 2023-02-21 by gnuton