gnuton/asuswrt-merlin.ng #339: Cannot add as mesh node to rt-ax86u with 388.1 gunton and rt-ax82u

cpuwaiy, 2023-02-06:
**Router Model Affected**
RT-AX82U

**Firmware Version Affected**
388.1gnuton

**Is this bug present in upstream Merlin releases too?**
N/A

**Describe the bug**
Cannot add rt-ax82u to rt-ax86u with merlin firmware.  Had to reset both routers to asuswrt to link wit aimesh.

**To Reproduce**
Updated rt-ax86u to merlin 388.1
Updated rt-ax82u to merlin 388.1 gnuton
Add rt-ax82u as mesh node to rt-ax86u

**Expected behavior**
Rt-ax86u would discover the rt-ax82u when trying to add it as a node

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

gnuton, 2023-02-07:
In order to setup aimesh you must connect the routers with an Ethernet cable. Then after the first setup you can remove it and they will connect via wireless. This is a known issue of my firmware. 
Have you tried this?

cpuwaiy, 2023-02-07:
I did the reverse.  I have a wired backhaul, will that be a problem?

gnuton, 2023-02-07:
not at all. wired backhaul works all the time.
Again the only problem is that setup doesn't work through wireless. I do still need to find the reason... but the rest works as it should.

cpuwaiy, 2023-02-07:
great. Thanks for the fast response!

gnuton, 2023-02-07:
I will close this support request

closed 2023-02-07 by gnuton