gnuton/asuswrt-merlin.ng #340: Add nat6(napt66) connection type for the ipv6 connection type

yk00365, 2023-02-06:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when I have a modem like tewa-707g, its router mode's ipv6 address cannot suitable for the passthrough mode. But the nat6 mode can suitable, maybe asuswrt-merlin must support dhcpv6 mode,maybe I think you can repair and add it.

**Describe the solution you'd like**
nat6(napt66)

**Describe alternatives you've considered**
For example,padavan has fix it.

gnuton, 2023-02-07:
Can you please show me the fix you are talking about? Which router model?

yk00365, 2023-02-07:
> Can you please show me the fix you are talking about? Which router model?

Sure. The napt66 is a nat6 mode, I also have a router, it runs for a firmware called padavan, But I surprised its native mode can suitable for the modem called tewa-707g. Unfortunately, the asuswrt-Merlin is not suitable for the tewa-707g's router mode ipv6 address.
I am not an English native speaker. The two pictures can show the bug.
[image]
[image]

The router mode is rt-ax82u,tuf-ax3000,rt-acrh17.
It is the same bug.

gnuton, 2023-02-08:
IPV6 should be already supported by the rt-ax82u.

There were some issues with it for some ISP but as far as I know that bug has been resolved.
The ISP affected were in australia and japan.. but now i have been told ipv6 works for those ISPs too since the last version.
There is a guide here and passthorough is supported too
https://www.digitalcitizen.life/ipv6-asus-router/

yk00365, 2023-02-11:
> RT-AX6u 应该已经支持 IPV82。
> 
> 对于某些ISP来说，它存在一些问题，但据我所知，该错误已解决。受影响的ISP在澳大利亚和日本。但是现在我被告知，自上一版本以来，IPv6也适用于这些ISP。这里有一个指南，也支持通过 https://www.digitalcitizen.life/ipv6-asus-router/

But，some modem like tewa-707g is based on openwrt, if the router run openwrt,and modem as a first router and open ipv6 connection, my ISP is China telecom, it selected ipv6 as dhcpv6 mode, like padavan can connect is, but asuswrt is not get the ipv6 address, if I open the passthrough mode, I am connecting the router with the ssh, when I type the ifconfig and press enter, I feel be shocked at the result only show for a local ipv6 address like fe80 ,but not see a 240e  ipv6 address, it is not recieve and ipv6 address. So I am not can use the ipv6 connection. When I used the openwrt on the rtacrh17, I feel shocked too, it not selected any selections, but can alredy recieve the dhcpv6-pd, already can used, so I feel angry with asus and write an e-mail to ASUS, also even if the company tells me waiting for the updates, but is not be repired.
So, could you can fix the bug? Many thanks.