gnuton/asuswrt-merlin.ng #341: Slow speed with other VPN's when using Wireguard

conman33158, 2023-02-07:
**Router Model Affected**
Models: (RT-AX82U)

**Firmware Version Affected**
RT-AX82U_388.1_0-gnuton1_puresqubi

**Is this bug present in upstream Merlin releases too?**
Yes and with official release

I'm a problem on BT 900mb connection. It only does it with SurfShark VPN. When Wireguard VPN is switched on, on my Asus RT-AX82U even though no clients are connected it gives me no more that 320mb speed Via SurfShark VPN but when not using Wireguard the SurfShark Wireguard protocol it's around it's normal speed of 890-930mb. If I use the free version of Proton VPN via Netherlands server I get around 800mb which is really strange so it seems to be only SurfShark I'm having the problem with.

I wanted the Wireguard on the Asus as a backup as I use PiVPN on a Raspberry Pi but liked the idea of having Asus Wireguard for a failover. I guess that's not gonna happen unless I stay capped at around 300mb when using SurfShark at home on any of my devices. I could always switch on Asus Wireguard when going away and then back off when I'm home but seems a bit of a hassle to be honest.

gnuton, 2023-02-07:
If I got it right the speed drop comes from the fact that enabling wireguard disables the hw acceleration.

conman33158, 2023-02-07:
> If I got it right the speed drop comes from the fact that enabling wireguard disables the hw acceleration.

Is there a way to disable HW acceleration my friend?

gnuton, 2023-02-07:
HW Acceleration is automatically switched off when you run wireguard on the router

conman33158, 2023-02-07:
> HW Acceleration is automatically switched off when you run wireguard on the router. HW acceleration is what makes the connection faster.. so you want to have it on. do not run wireguard on your router if speed is a concern, the limitation is in the router hw.

Ok man, no worries 👍

aherped, 2023-02-11:
> HW Acceleration is automatically switched off when you run wireguard on the router. HW acceleration is what makes the connection faster.. so you want to have it on. do not run wireguard on your router if speed is a concern, the limitation is in the router hw.

Hello, I want to put here this threath https://www.snbforums.com/threads/why-does-wireguard-slow-down-the-entire-wan-lan-speed.81636/page-2#post-811997

It seems not to be a HW problem or limitation as they said that: "Thanks for all replies! With the new official 3.0.0.4.388.22068 firmware the problem seems gone! I get full 1 Gbps from clients outside the VPN and 400-500 Mbps from clients inside. So in the end it didn't seem hardware related."

It's possible to implement this?

Thank you

conman33158, 2023-02-12:
> 

We are talking about using a separate VPN like SurfShark when Wireguard is switched on via the Asus router settings, not the actual speed from Asus Wireguard itself. 

What you mention is NOT relevant to the topic my friend.

conman33158, 2023-02-12:
> Hello, I want to put here this threath https://www.snbforums.com/threads/why-does-wireguard-slow-down-the-entire-wan-lan-speed.81636/page-2#post-811997
>
> It seems not to be a HW problem or limitation [...]

We are talking about using a separate VPN like SurfShark when Wireguard is switched on via the Asus router settings, not the actual speed from Asus Wireguard itself.

What you mention is NOT relevant to the topic my friend.

aherped, 2023-02-12:
> We are talking about using a separate VPN like SurfShark when Wireguard is switched on via the Asus router settings, not the actual speed from Asus Wireguard itself.
>
> What you mention is NOT [...]

Hello conman33158, 

You are right, my apologies, I misunderstood the initial problem. 
I'm going to open a new topic with my problem. 

Thanks for the advise.

gnuton, 2023-02-16:
Asuswrt-Merlin's WireGuard implementation is from Asus, so if this has been fixed by asus somehow then it will come with the next merlin release.