gnuton/asuswrt-merlin.ng #342: Wifi issues? cannot ping router or others for a few seconds.

bobyang3, 2023-02-09:
**Router Model Affected**
Models: TUF-AX5400

**Firmware Version Affected**
388.1.0-gunton1_tuf

**Is this bug present in upstream Merlin releases too?**
unknown

**Describe the bug**
I don't see my computer or phone wifi get disconnected. However, the network connection is dropped for a few seconds randomly. it happens about 4 times per day after I upgraded from 386 to 388 two days ago. I am pinging the router and other machines, suddenly, ping gets no response for a few seconds, other machines are losing pings as well. I notice it because I RDP to other machines.

I don't see anything really catch my eyes in system log.. perhaps I saw a lot of following 

Feb  9 09:14:27 wlceventd: wlceventd_proc_event(469): eth5: Deauth_ind E8:2A:EA:xxxxxx, status: 0, reason: Unspecified reason (1)
Feb  9 09:14:27 wlceventd: wlceventd_proc_event(505): eth5: Auth E8:2A:Exxxxx, status: Successful (0)
Feb  9 09:14:27 wlceventd: wlceventd_proc_event(534): eth5: Assoc E8:2Axxx, status: Successful (0)
Feb  9 09:14:59 wlceventd: wlceventd_proc_event(469): eth5: Deauth_ind E8:2A:xxxxx, status: 0, reason: Unspecified reason (1)
Feb  9 09:14:59 wlceventd: wlceventd_proc_event(505): eth5: Auth E8:2A:EAxxxxx status: Successful (0)
Feb  9 09:14:59 wlceventd: wlceventd_proc_event(534): eth5: Assoc E8:2A:xxxx, status: Successful (0)

**To Reproduce**
Steps to reproduce the behavior:
unknown

**Expected behavior**
not  losing connection

**Screenshots**
provide later.

bobyang3, 2023-02-10:
I tried 388.1.0-gunton1 about 5 hours ago. it seems like it is fine so far. I don't have ping lost anymore.

bobyang3, 2023-02-10:
it seems like it is happening. I have auto reboot every other day at 3:xx am. it rebooted this morning, and after 6 hours of online, it starts doing the same thing again about losing connection once a while, it happens 4 times so far.

rlfrank165, 2023-02-10:
I just want to reiterate that everything being reported with the various
Gnuton versions I saw with the various Asus official 388 releases and are
being complained about on various Asus community bboards.

For now I have given up on anything Asus 388 based and gone back to 386
based versions since I needed a stable network for a while. Life is back to
reliable in the 386 world.

I have tried to engage Asus official tech support, but sadly they don't
seem that interested, at least at the levels I'm allowed to talk to.  I'm
hoping that someone in the Mirlin community has higher level contacts that
can convince Asus that they have a real problem that is seriously affecting
their reputation as a quality networking provided.

Randy Frank

bobyang3, 2023-02-10:
@rlfrank165 Thanks. I think it could be related to channel changes or I have Dual-Band Smart Connect (2.4Ghz and 5GHz) but I am not sure. I am going to give a try for a few more hours, if it happens, I will switch back to 386 as well because I also need a reliable network which I DRP to different machines a lot.

core1983, 2023-02-13:
Hello,
I have a similar problem when downloading a larger file from the web, a network error appears after a minute or two.