gnuton/asuswrt-merlin.ng #344: Slow Internet Speed if Wireguard VPN is enabled

aherped, 2023-02-12:
**Router Model Affected**
Models: RT-AX92U

**Firmware Version Affected**
388.1_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
Don't really know, I think yes.

**Describe the bug**
When I activate Wireguard VPN, even if no client is connected, my internet speed inside my network drops from 940mbps to 450mbps with speedtest. If I disable Wireguard VPN I get 940mbps again with no problem.

I read about an optimization of Wireguard in newer versions of firmware but on other models.

https://www.snbforums.com/threads/why-does-wireguard-slow-down-the-entire-wan-lan-speed.81636/page-2#post-811997

It seems not to be a HW problem or limitation as they said that: "Thanks for all replies! With the new official 3.0.0.4.388.22068 firmware the problem seems gone! I get full 1 Gbps from clients outside the VPN and 400-500 Mbps from clients inside. So in the end it didn't seem hardware related."

Thank you

lhayati, 2023-02-14:
I was just about to post about a throttle bug on my XT8 (RT-AX95Q). My speed moving from 386 to 388 dropped from 940mbps to about 300mbps. I read your post, turned off the wireguard server and immediately it bumped back up to 940mbps! Temporarily solution I'll setup a wireguard server on a raspberry pi behind the router.

gesa, 2023-03-22:
This is because NAT acceleration gets disabled when Wireguard is in use. Something about a proprietary broadcom implementation of flow cache not being compatible with wireguard. You can confirm by checking out Tools > Network > HW acceleration. Speeds are limited to approx 350mbps without hw acceleration.

The good news is that [388.2 will use a new feature implemented by Broadcom/Asus which will allow non-WG traffic to still benefit from NAT acceleration](https://www.snbforums.com/threads/slow-wan-with-vpn-director-turned-on.84017/post-828075), the bad news is it’s unclear when that’s released.

mdburkey, 2023-04-23:
Looks like 388.2 Merlin mainline is out...hoping for a gnuton fork update for the XT8!

seirdotexe, 2023-06-14:
I'm facing this exact issue on my TUF-AX5400, but with Asus' default firmware. I'm on 3.0.0.4.388_23285. Any ideas?