gnuton/asuswrt-merlin.ng #345: AiMesh, not all wifi settings not getting passed back to the slave node

DJE007, 2023-02-12:
**Router Model Affected**
Master RT-AX88U / Slave DSL-AX82U

**Firmware Version Affected**
388.1

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
Not all AiMesh wifi settings not getting passed back to the slave node

RT-AX88U-9C90:/tmp/home/root# nvram getall | grep -i rateset
**size: 72454 bytes (58618 left)
wl0_rateset=ofdm
wl1_rateset=ofdm
wl_rateset=ofdm**

RT-AX88U-9C90:/tmp/home/root# nvram getall | grep -i ofdma
size: 72454 bytes (58618 left)
rc_support=mssid 2.4G 5G update usbX2 switchctrl manual_stb 11AX pwrctrl WIFI_LOGO nandflash movistarTriple wifi2017 app ofdma wpa3 reboot_schedule ipv6 ipv6pt PARENTAL2 loclist dnsfilter dnspriv dualwan pptpd openvpnd utf8_ssid printer modem webdav rrsut cloudsync media appnet timemachine hdspindown diskutility dnssec usb_bk frs_feedback dblog email bwdpi wrs_wbl ookla HTTPS letsencrypt ssh vpnc vpn_fusion repeater psta wl6 wifi_tog_btn user_low_rssi tcode usericon stainfo cloudcheck realip lacp wanbonding wtfast ifttt alexa ipsec_srv mumimo netool cfg_sync no_finiwl fupgrade afwupg betaupg amas bcmwifi bcmhnd mbo conndiag eula proxysta iperf3 gn_wbl amazon_wss account_binding gameMode wireguard ftp_ssl acl96 dhdlog dis11b smart_connect
**wl0_ofdma=3
wl1_ofdma=3
wl_ofdma=3**

**Slave Node**

DSL-AX82U-1058:/tmp/home/root# nvram getall | grep -i rateset
**size: 86909 bytes (44163 left)
wl0_rateset=default
wl1_rateset=default
wl_rateset=default**

DSL-AX82U-1058:/tmp/home/root# nvram getall | grep -i ofdma
rc_support=dsl vdsl spectrum mssid 2.4G 5G usbX1 update switchctrl manual_stb 11AX pwrctrl nandflash wifi2017 app ofdma wpa3 defpsk reboot_schedule ipv6 ipv6pt s46 PARENTAL2 loclist dnsfilter dnspriv dualwan mtwancfg pptpd openvpnd utf8_ssid printer modem webdav rrsut cloudsync media appnet timemachine hdspindown diskutility dnssec usb_bk frs_feedback dblog email bwdpi wrs_wbl ookla HTTPS letsencrypt ssh vpnc vpn_fusion wl6 user_low_rssi tcode usericon jffs2 cfg_wps_btn stainfo cloudcheck realip alexa ipsec_srv mumimo netool cfg_sync no_finiwl fupgrade afwupg betaupg amas bcmwifi bcmhnd mbo conndiag eula proxysta iperf3 mswan account_binding gameMode wireguard ftp_ssl acl96 hnd_ax_675x dhdlog dis11b smart_connect
size: 86910 bytes (44162 left)
**wl0_ofdma=3
wl1_ofdma=3
wl_ofdma=1**

DJE007, 2023-02-12:
work around ssh into the slave and,

nvram set wl0_ofdma=3
nvram set wl0_rateset=ofdm
nvram set wl1_ofdma=3
nvram set wl1_rateset=ofdm
nvram set wl_ofdma=3
nvram set wl_rateset=ofdm

nvram commit && reboot

gnuton, 2024-03-17:
Is this a problem also in the stock firmware?

DJE007, 2024-03-17:
I have moved on to a unifi ap mesh solution