gnuton/asuswrt-merlin.ng #346: Memory consumption 388.1_0-gnuton1

Rovasteen, 2023-02-15:
TUF-AX5400 using 388.1_0-gnuton1.  When rebooted, the router is using 52% of RAM.  The RAM continues to be consumed over time.  At 21 days uptime, the RAM consumed is 68%.   Seems like a memory leak.  I noticed that Merlin has updated components in his 388.2 alpha Builds.   Would be interested to try an updated Gnuton build with updated components to see if there is a positive impact on memory consumption.

lhayati, 2023-02-15:
RAM is designed to be used, unused RAM is wasted ram. If it reaches so high that it starts to impact performance then I'd classify that as an issue, but 68% is well within typical use so I wouldn't worry. My XT8 is sitting at 70% right now.

Nehaxak, 2023-02-15:
It's been discussed elsewhere on the issues here recently but it's most likely related to the following I'll copy/paste below, and which Asus recognise and apparently have fixed so yes, maybe the current alpha from merlin has included the code to fix this. I've the same router and the same problem btw. There's some script workarounds available in the link below you could try while waiting on new builds that include the fixes from Asus if it's affecting you badly...

---

Post in thread 'Out of Memory Errors 388.1 / No space left on device' https://www.snbforums.com/threads/out-of-memory-errors-388-1-no-space-left-on-device.82406/post-819862

"Asus confirmed that the conn_diag database issues were already fixed on their end, and should therefore be included in the next code release."

---

Rovasteen, 2023-02-15:
Thanks @Nehaxak that is useful information!!!  It is a real issue. On the 386 builds, I could run the router for months and never get much above 56% memory consumption.  With same usage on 388 builds, up to 68% memory used after just 3 weeks.

gnuton, 2023-02-16:
I will close this because this has been already analyzed.
@Rovasteen feel free to open this ticket again once you have profiled the application that is leaking memory.

@Nehaxak the no memory left on the devices is a different story. That's related to verbose loggigng which fills the disk and not the RAM.

closed 2023-02-16 by gnuton