gnuton/asuswrt-merlin.ng #348: 388.1 slow wan speeds after consistent downloading

ZorianNL, 2023-02-19:
**Router Model Affected**
Models: RT-AX82U

**Firmware Version Affected**
388.1

**Is this bug present in upstream Merlin releases too?**
Not sure, there is none for this model.

**Describe the bug**
If you download something for a while (like something big), the internet speed drops to a crawl until you reboot the router.
I have an 1000/1000 Mbit connection, it starts out fine at ~930 Mbit, then after a few minutes it crawls to about 20 Mbit or lower, while bringing the ping from 3-4 ms to >1000 ms. It will continue to become slower and slower until you reboot, then rinse and repeat. If you do not reboot, eventually the internet just cuts out (slower than dial-up speeds, which basically is the same as nothing).
Already tried factory resets, reflashes, flashed back to 386 and confirmed it being fine, then back to 388 and see the issue pop up again.

**To Reproduce**
See above paragraph.

**Expected behavior**
Not crawl to a snail speed, previous 386 stable build is fine and doesn't have this issue.

**Screenshots**
N/A

jackbeagley, 2023-02-22:
I have a very similar issue with my AX82U on the same firmware, for my wireless clients it can be resolved temporarily by turning the client's wifi on and off.

Estaira, 2023-05-11:
I have the same issue on AXE95Q router.