gnuton/asuswrt-merlin.ng #349: TUF-AX3000 v1's Assassin mode, 4T4R.

Jacky10X, 2023-02-19:
There is a mode call Assassin mode(刺客模式). It seems like it only appear to China model (CN/01). It unlock the 5Ghz 2T2R to 4T4R. 
Once you enable to the Assassin mode
it seem it shows Australia in wireless setting But it is still broadcasting CN as wireless country code.
And it seems like it only available as the main router, not the Aimesh node mode.

And i saw there is a firmware that you can unlock 4T4R in all region.

My feature request is 
Could this firmware add an option to unlock 4T4R in all region (like the assassin mode but it boradcasting as the region you choose)?
Also, make it work as Aimesh node while you can enabling 4T4R too?

I am sorry if i am asking too much.

Jacky10X, 2023-02-24:
@gnuton Could this feature/function possibly achieve?

c2707637, 2023-03-08:
Useful function! 🙏

Jacky10X, 2023-07-28:
> Search for SWRT, it unlock 4T4R, even on global version.

SWRT might have some safety & security issues. you can search for it on the Internet.

Jimmy8881, 2024-04-25:
> Search for SWRT, it unlock 4T4R, even on global version.

anything like this possible on the V2 ? their page lists the v2 but theres no download link.

alfky12, 2024-09-07:
Hi @gnuton, bringing up this request again, can you open assassin mode for all regions? Thanks.

Jacky10X, 2024-09-12:
> > Search for SWRT, it unlock 4T4R, even on global version.
> 
> anything like this possible on the V2 ? their page lists the v2 but theres no download link.

This is not possible on the V2 because the 5Ghz chip is different.

alfky12, 2024-09-12:
@gnuton  I can bring up and turn on the assassin mode toggle using the init-start script, and it persists even when the router is rebooted, but 4T4R is not achieved and the signal output power is unchanged.

I don't know if this is a system security measures for non-china region or a bug in the gnuton firmware. Have you tried your firmware can activate assassin mode properly on tuf-ax3000 china region?

gnuton, 2024-09-13:
@alfky12 are you in router mode? Have you tested the other firmware that @Jacky10X is proposing? That may contain this "fix" along with backdoors :) 
If we can document how to properly make this working that would be awesome. 
I would also suspect that not all hardware will be working with this hidden feature. I have not checked the code for this yet, but usually hw settings are part of low level drivers which I cannot touch... But if someone like @Jacky10X is more keen to find a way to get this working I can make sure the firmware doesn't stop you to do so if there are changes to be made

gnuton, 2024-09-13:
Gotcha. So to enable this feature, the prerequisite is not on the firmware but rather in the custom CFE. That s something I usually I do not mess up with since it can hard brick your router.
Since flashing my firmware won't touch the bootloader, you can try to flash the first firmware and later mine. Check the bootloader version with the first fw and later after flashing mine. Then you can test again if changing locale enable the feature.

Jacky10X, 2024-09-13:
> @Jacky10X I do not think I will enable this can be enabled by default since it will make the firmware illegal. I think the best option is to make a script to enable this as add-on that can be downloaded throug amtm.

Understood. After some research, It seems that to enable 4t4r on RT-AX58U v1 (not v2!). you can nvram set 1:sw_txchain_mask to 0xf to enable 4T4R.(not confirm) I don't have RT-X58U v1 so I can't confirm it. For TUF-ax3000, this doesn't work. It is still 2T2R or you can say 2T4R.

alfky12, 2024-09-13:
Hi @qqw1231 sorry for mentioning you here. I see you have the chinese version of tuf-ax3000, does assassin mode work fine on gnuton firmware?

Jacky10X, 2024-09-14:
After some digging, I found the /www/device-map/router.asp limits the assassin mode to CN only. But if we change the CN to another region or remove the limit. It might work for all regions.  (not confirmed)

gnuton, 2024-09-16:
@Jacky10X you can edit that page and mount it to check if what you are saying is correct. asp pagaes are usually sending info to to the HTTP server which is written in C. As long as there is code to be modifed that's not an issue.
If oyu like you can also do some changes, push the changes to github and let it create the images for you to test.
If you wanna show code or you require some guidance feel free to ping me

Jacky10X, 2024-09-18:
After forcing to enable assassin mode, it does not work. I think it is related to cfe. The Chinese model's cfe is different from others. I believe it requires China mode TUF-ax3000 cfe or other custom cfe.

Jacky10X, 2024-09-24:
@gnuton I think I enabled the 4T4R. The firmware limit only the cfe's region (CN) to enable the 4T4R.
The first method is you need to modify the territory_code from cfe. Otherwise, it won't work. but It has limited options. [e.g. only available as main router, etc...]
The second method is to modify the firmware with cfe.