gnuton/asuswrt-merlin.ng #350: DSL-AX82u - possible DNS issue? 388.1_0

f1ux-capacitor, 2023-02-21:
**Router Model Affected**
Models: DSL-AX82U

**Firmware Version Affected**
388.1_0

**Is this bug present in upstream Merlin releases too?**
Not sure

**Describe the bug**
Wired/wireless devices randomly lose connection. 
e.g starting a stream on appletv (Wi-Fi 5ghz) all lan/wan connectivity on the device is lost until I toggle Wi-Fi. Same occurs on my iPhone. 
on my MacBook Air server (Ethernet) running a VM of homeassistant, the VM loses  connectivity until it’s restarted. While it happens the MacBook is still accessible on the network and the MacBook can access the VM via localhost ok. 

**To Reproduce**
Steps to reproduce the behavior:
Happens randomly. With Wi-Fi devices,toggling Wi-Fi resolves every time. 

**Expected behavior**
Stable connection

**Screenshots**
N/a

gnuton, 2023-02-21:
can you pleae factory reset the router and check if this happens again?
please do not restore backup settings

f1ux-capacitor, 2023-02-25:
I was still getting the same issue after a factory reset. 
live dropped Wi-Fi back from WPA3 to WPA2 and so far none of my Wi-Fi devices have had issues. 
my server connected to LAN does though once a day. 

I’ve also noticed the built in speed test is much slower. E.g it gets around 300-600Mbps , yet on laptop to same server it’s always over 800. 
previously the router speed test always hit max speed. 

I’ve checked temps and they seem a little high for a Room that’s 18c

2.4ghz:51c
5ghz:  48c
CPU: 68c