gnuton/asuswrt-merlin.ng #351: Random Reboots

JonWVBFL, 2023-02-23:
**Router Model Affected**
Models: RT-AX95Q (1 Router + 1 Node)

**Firmware Version Affected**
388.1_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
Did not occur previous version of 386.08_0-gnuton1

**Describe the bug**
The router randomly reboots during the day. Has been occurring since recent upgrade to 388.1. I have performed a factory reset on the system in hopes that would have stopped this from occurring.

Features I am utilizing (non-default):
- AIProtection
- Parental Controls - Web & Apps Filters; Time Scheduling
- Bandwidth Monitor & App Analysis
- Web History
- DNS Director
- Dual WAN

**To Reproduce**
Normal usage of router throughout the day, will randomly restart. I have pulled the system log and pruned out the crash log it provides and attached.

**Expected behavior**
To not reboot.

[Crash1.txt](https://github.com/gnuton/asuswrt-merlin.ng/files/10818881/Crash1.txt)

gnuton, 2023-02-24:
Hi,
thanks for reporting this.
The issue seems to be related to WiFi drivers on eth5
Please let me know if you still see this issues after factory reset.

hyprsnpr, 2023-02-24:
I had this same issue. I attached a external SSD to a USB port and enabled swap, it stopped the crashes.

Use to enable swap: [amtm](https://diversion.ch/amtm.html)

JonWVBFL, 2023-03-14:
> Hi, thanks for reporting this. The issue seems to be related to WiFi drivers on eth5 Please let me know if you still see this issues after factory reset.

@gnuton Sorry for the confusion, I did a factory reset before submitting the issue. It does seem to happen almost every day, I just noticed my uptime right now is at 1hr on the system. Sorry for the delayed response...i forgot to enable my notifications.

JonWVBFL, 2023-06-13:
@gnuton I've been running 388.2.2 since June 1 and it seems stable now, it was occurring almost daily! Thanks for the great work on this firmware!

closed 2023-06-13 by JonWVBFL