gnuton/asuswrt-merlin.ng #354: TUF-AX5400 froze list of clients after 2 weeks

Barmagler, 2023-02-26:
**Router Model Affected**
Models: (TUF-AX5400)

**Firmware Version Affected**
388.1-gnuton1_tuf

**Is this bug present in upstream Merlin releases too?**
No

**Describe the bug**
After 10-14 days, the list of clients and client status stops updating, the number of clients froze at 14 positions.
At the same time, 2.4 / 5 GHz clients connect without any problems.
I have 2 wired, 7 on 5 GHz and 9 on 2.4 Ghz clients.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to router more than 14 clients
2. After 2 weeks connect one more new client.
3. Try to update list of clients on "Network map"
4. Number of clients froze at 14 positions

**Expected behavior**
List of clients should be show correctly.

gnuton, 2023-02-27:
It feels to me this is browser issue. maybe using old cookie for authentication.
Please try to browse with incognio and clean your browser cache

Barmagler, 2023-02-28:
I tried from a different browser on the same computer and then tried from a laptop. The only thing I didn't do was not use the app. At the moment, the uptime is already 3 days, if the bug does not happen again in the near future, then this is definitely a browser.

In any case, I apologize for such a stupid treatment :(

Can I keep it open for week more?

mohamedth82, 2023-03-04:
I faced same issue with TUF-AX5400
clients list froze
I did not reboot my router since 1 month

Barmagler, 2023-03-04:
One week uptime at this moment so continue to watch situation.
On prev version gnuton uptime was more than 1 month and no problems with freeze list.

closed 2023-05-01 by Barmagler