gnuton/asuswrt-merlin.ng #355: custom DOT is set but dnscheck shows others too

damasko1, 2023-03-05:
In the RT-AX82U I set the nextdns DOT as unique server, but the site dnscheck.tools shows that I am using other dns resolvers, further to the nextdns which I set.

[image]

[image]

damasko1, 2023-03-08:
news after further investigation: I found cloudfare set in browser doh, and now I removed it, but m1ns.com still remains as a resolver.
it seems that the router setting "prevent client auto doh" does not work.

Nehaxak, 2023-03-08:
@damasko1

Turn the DNS Director on and set "Global Redirection" to "Router"

LAN > DNS Director > 

In here also you can set specific devices to be bound and use specific DNS, like SmartDNS services etc.,

Anyway, I've found that turning that on myself and setting the global redirection to router has helped with the NextDNS cli client doing what it's supposed to be doing and stopping phone apps in particular, like tiktok, from bypassing nextdns and using their own DNS service/in-built server. Everything just gets redirected to the router and then through the nextdns cli as it should be. 

Hope that helps, just a suggestion anyway, worth a try. 

Reboot your router after setting it btw, just to be on the safe side that the setting holds.

R3D-PLAN3T, 2023-03-19:
It seems that the manual DNS setting is causing the problem. I have to take DNS from my ISP (automatically) otherwise at some point I lose the internet connection and all devices in the network think that they are not connected to the internet, although the DSL router has internet, but because of Cloudflare manual DNS you get this error message.