gnuton/asuswrt-merlin.ng #356: Processor 3 Dormant in 388 Firmware

SgtParper, 2023-03-19:
**Router Model Affected**
Models: DSL-AX82U

**Firmware Version Affected**
388.1_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
I believe so.

**Describe the bug**
Updating firmware from 386 to 388 branch results in processor core 3 becoming dormant. 5Ghz wifi also became unpredictable. Core 3 also displayed same behaviour with stock 388 firmware after factory reset. 

**To Reproduce**
Steps to reproduce the behavior:
1. Update firmware to 388 branch

**Expected behavior**
Processor load to be spread over all 3 cores. Have downgraded to 386 branch, which has expected behaviour. Same behaviour on 2 different DSL-AX82U routers.

WillyTP, 2023-03-19:
Hi,
actually I'm experiencing this too.
I never noticed before but it's true.

DSL-AX86U.

Kind regards

SgtParper, 2023-03-21:
The typical behaviour is for all 3 cores to share the load in my experience. These pics were of my processor load whilst it was just reinitialising after a reboot.

[image]
[image]

As you can see the load is spread fairly evenly and without favouring any core (or cores) in particular. This is historically the behaviour that I expected and is the case with 386.08 firmware. If there has been some change to the code that assigns particular functions to particular cores in 388.1, it would be interesting to read the mapping of these.

(Sorry that the pics are so big.)