gnuton/asuswrt-merlin.ng #358: [AX92U][388.1]Connection down after a few days

nnciprian, 2023-03-22:
**Router Model Affected**
Models: RT-AX92U

**Firmware Version Affected**
388.1

**Is this bug present in upstream Merlin releases too?**
N/A

After a few days of uptime, in this case 5 days router is disconnected from the internet and I have to reboot to connect again. Internet LED does not go red (i.e. stays white).
In system log I see some memory outages sometimes and no free space left on the device on some mounts;.
I attached system log for today's event since 00:00AM. Link down I believe it happened around 5 am;

Some screenshots attached;
[image]
[image]
[syslog_git.txt](https://github.com/gnuton/asuswrt-merlin.ng/files/11037142/syslog_git.txt)

nnciprian, 2023-06-15:
after connection diagnostics issue still persists

Jun 14 16:23:51 dnsmasq-dhcp[16036]: failed to write /var/lib/misc/dnsmasq.leases: No space left on device (retry in 60 s)

gnuton, 2023-06-15:
Please update to latest stable

gnuton, 2023-06-15:
duplicate of  #334

closed 2023-06-15 by gnuton

nnciprian, 2023-06-17:
updated to 388.2_2_0 ;
Monitoring