gnuton/asuswrt-merlin.ng #362: Unable to revert back to stock firmware

Chilly1328, 2023-03-24:
**Router Model Affected**
Models: RT-AX82U

**Firmware Version Affected**
388.1

**Is this bug present in upstream Merlin releases too?**
Unknown

**Describe the bug**
Try to upload the latest firmware from Asus and go to manual firmware update, the system would report "firmware upgrade unsuccessful. this may result from incorrect image or error transmission. please check the version of firmware and try again." and refuse to update.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Firmware Update' under 'Administration'
2. Click 'Manual update'
3. Select stock firmware
4. See error

**Expected behavior**
Stock firmware would be applied.

gnuton, 2023-03-25:
You have a V2 and you have flashed a V1 firmware on it. 
There is no way to brick the router, so do not be worried about it.

If flashing doesnt work from the UI please use the Recovery tool provided by asus

r7rbb, 2024-12-12:
Hi guys 
i am not sure that i am writing in correct page, but this only what i found close to my question.

i have asus as82u  firmware 388.2_2-gnuton1.
i intend to sell the router and want to restore to Asus wrt.
Just downloaded latest asus firmware flashed manualy ( all was done like usial) but when rebooted after flashing , i stay thensame  388.2_2-gnuton1. no changes. 
  i did hard reset. but the same.
Could you ls advise what i did wrong?

gnuton, 2024-12-12:
@r7rbb it should work. make sure you have a stable connection. if it still doesn't work then this will defintely work
https://www.asus.com/support/faq/1000814/