gnuton/asuswrt-merlin.ng #365: 388.2 status updates

gnuton, 2023-03-27:
Here you can find updates about the status of the next 388.2 release

Planned in May.
Current status:
- GPL Collection [IN PROGRESS]
   - ASUS is still preparing these for me
     - DSL-AX82U GPL
     - RT-AX95Q, RT-AXE95Q [at moment there are issues here]
- GPL Merges [IN PROGRESS]
- Pre-Alpha testing
- Alpha release
- Beta release
- Public release  [?? May ?? ]

CPngN, 2023-03-27:
What are the issues with RT-AXE95Q / ET8 ?  I desparately need to get off stock, it is garbage when everyone is home, even with wired backhaul!

gnuton, 2023-03-28:
@CPngN it doesn't build.

neil1962, 2023-03-29:
I was just wondering when a Alpha/Beta version of 388.2 may be available?

CPngN, 2023-03-29:
Oh my bad, when I asked I didn't notice that this was 388.2, and that .1 was already out.  

Good luck with the build! 

I will probably start my Merlin anointment with .2 unless I can't take it any more. I had to reboot both nodes today physically and there are still 3 IoT switches that won't get back online. Tired of this!

gnuton, 2023-03-31:
Today I have received the other GPLs. X/ET8 have different version -_-
I will merge those in the next days

dacootmeister, 2023-03-31:
> Today I have received the other GPLs. X/ET8 have different version -_- I will merge those in the next days

I'm very excited to check out new functionality! Thanks @gnuton :-)

Rovasteen, 2023-03-31:
Would be great if an alpha or beta build could be released as Merlin did for routers where the GPL 388_22525 is available as it solves a lot of lingering problems.  Thank you!

micron888, 2023-04-01:
excited to see new release coming.
I am still using 386.07_2-gnuton due to GUI and RAM issues on 388

CPngN, 2023-04-02:
GUI and RAM issues on 388?! Oh no! I'm on 386 and two ET8's just cannot keep up with 70 devices with 5 users streaming and playing games. I was hoping 388 may solve the horrible performance I get after a few days.  
It's feeling like I just need to sell this crap and get something more grown-up.. grr

neil1962, 2023-04-03:
> GUI and RAM issues on 388?! Oh no! I'm on 386 and two ET8's just cannot keep up with 70 devices with 5 users streaming and playing games. I was hoping 388 may solve the horrible performance I get after a few days. It's feeling like I just need to sell this crap and get something more grown-up.. grr

So sell it and just move on, really!

Nehaxak, 2023-04-03:
You're not being very helpful either there Neil with that fanboish reply to the guy. People are rightly frustrated at Asus for their extremely poor firmware over the last few months and slow updates and slow support.

Don't see him as having a go at Gnuton either, just expressing, rightfully, frustration at Asus but sure the rest of us all know Asus couldn't really care less either way.

Personally myself, I've taken the Asus routers off the network over the weekend and gone with Ubiquiti, I've just had enough of ASUS poor firmware and wifi randomly dropping with zero info in any logs explaining why.
I'll check back in, in maybe a year, see if they've bothered to care because at the moment, Asus has just reverted to more poor quality Chinese ewaste.

Thanks Gnuton for all the hard work regardless, not your fault Asus are wasters.

CPngN, 2023-04-05:
When I moved in 2020, my old AC5300 wasn't cutting it. It couldn't reach half the house well enough to maintain stable connections. I jumped at what appeared to be the best 6e option at the time, with a UI I was familiar, and mesh compatibility with the AC5300.  But the best they have ever performed (without randomly dropping devices or causing grief with video and gaming streams) was with the pre-rc3 test build that was "available" last year.  Even then, it only lasted a few days before I had to reboot.  
I'm looking to move to 388.x and solve some of these issues, but given a couple of comments here I was worried that it still suffers from the same dropout and low bandwidth issues.  I'm definitely not against trying it, I'm preparing to actually, but was put off when I read it may be worse than the 386 base.  
If you believe 388.1 or .2 are still better than 386, I'm all in!

neil1962, 2023-04-06:
@gnuton any update on the 388.2 builds?

Rovasteen, 2023-04-06:
I have the same question.  Running Merlin 388.2_beta2 on another router and it is great.  Would love to have this build on my TUF-AX5400 w/Gnuton!

neil1962, 2023-04-07:
Ya it's a little disappointing just how quiet interns of a status update and or 388.2 Alpha at least being available by now.

LarryRosen108, 2023-04-07:
You do know this is a volunteer effort? Be happy we get better firmware.

neil1962, 2023-04-07:
> You do know this is a volunteer effort? Be happy we get better firmware.
> 
> 
> > Ya it's a little disappointing just how quiet interns of a status update
> > and or 388.2 Alpha at least being available by now.

Really, of course I am aware of that, I am just commenting on the normal high standards to which gnorton has as aspired to before.

gnuton, 2023-04-08:
@neil1962 is right I should have made alpha available by now. It's not available because:
1.  I am travelling and I do not travel with a case of routers
2.  I am still facing issues with the SDK of the TU-F-AX3000
3. I got last images last week.

If you like I can send pre-alpha images for gathering feedback to selected people as long as you underestand that:
1. pre-alpha is untested and can soft brick your routers -> TFTP or Asus recovery tool may be required if something goes wrong
2. it's untested and it may have stil severe bugs

In any case I have still commited to get the 388.2 out in May.

Rovasteen, 2023-04-08:
I would like a pre-alpha image and to provide feedback. I am good with that.

neil1962, 2023-04-08:
Hi,

I am willing to, give it a try, I am using a rt-ax82u, enjoy your trip, and
stay safe.

neil1962, 2023-04-09:
Thankyou just now loaded it up to my rt-ax82u without an issue over the top of 388.1. Will now monitor and advise. Thanks gnuton for taking your time to provide this

neil1962, 2023-04-09:
Just another big thanks gnuton

Rovasteen, 2023-04-09:
Ditto. Will load 'er up and test.  Looking forward to the new GPL base.  Thanks @gnuton for preparing the image.

micron888, 2023-04-09:
dirty upgrade from 386 to  TUF-AX5400_388.2_beta1_puresqubi.w too.
[image]

Rovasteen, 2023-04-10:
So far 388.2_beta1 seems very stable on my TUF-AX5400.  Did a clean upgrade as I had some Wireguard remnants I wanted to clear out.

dacootmeister, 2023-04-10:
hi @gnuton I'd like a try as well with the alpha fiemware for the ZenWiFi_XT8 (base model: RT-AX95Q). Thanks for your efforts :-)

micron888, 2023-04-10:
after 10hours, flooded with cfg_server, 
cannot access web gui
[image]

micron888, 2023-04-10:
@Rovasteen  you have aimesh node?

Rovasteen, 2023-04-10:
No. I am using the router as a standalone node.

devdev999, 2023-04-10:
Flashed 388.2_beta1 on RT-AX82U and reset, with stock ax58u as AiMesh node, so far so good. Will continue to monitor.

Rovasteen, 2023-04-10:
@micron888 do you get this with other builds or just this 388.2_beta1?  Are you using guest networks?  If yes, which ones as Asus has "rules" about which guest networks can be used for varying purposes?

micron888, 2023-04-10:
@Rovasteen using guest 1, aimesh all nodes. original on guest2, turn to guest 1 after having some issues

neil1962, 2023-04-11:
Have been on 388.2 on my RT-AX82U for a couple of days now and all is very stable. I am not using anything special e.g., QoS, VPN, guest networks, no Trend Micro tools and no parental controls.  I am using in it AiMesh setup with 3 nodes right now, 2 which are connected via Ethernet and 1 via Wi-Fi.  I have had no disconnections, and log is clear, GUI is fast. RAM good and steady setting at around 52%.. Will contiue as is and report here, if anything appears.

Rovasteen, 2023-04-11:
Ditto. Also stable for me on TUF-AX5400.  Looking to see what the RAM usage goes to.  Currently at 54% after 1 day.

@micron888 Asus has made some changes in their guest network code, but this may still be correct:
Guest Network 1: uses 192.168.101.x and 192.168.102.x IP range, optimized for AIMesh and does not support IPv6
Guest Network 2: uses same IP range as main router, works with IPv6
Guest Network 3: uses same IP range as main router, works with IPv6, but has a lot of excess code due to Alex and IFTTT

I am not using AIMesh, so I use Guest Network 2 because I am using IPv6.  Asus has made some fixes to their guest network code, but I think this is still accurate.

micron888, 2023-04-11:
@Rovasteen  thanks for sharing. 
I follow instruction to set a static IP for Aimesh node, so far seem to work well
Running for 2hrs without any issues. will update again
 https://www.snbforums.com/threads/xt8-troubleshooting-log-cfg_server-possibly-an-issue.82445/
to s```

YameYukana, 2023-04-11:
using it on my ax5400 router for 16 hours. so far so good its really stable for me
gui is really fast and responsive
im using QoS and Wireguard VPN and they work as expected. maybe? gonna continue monitoring it
Also, I can’t see the button to enable Traffic Analyzer - Statistic

im trying guest network and ai protection now. ill update later if i see any problems with it

micron888, 2023-04-11:
> Anybody having issues in terms of connectivity with nodes, best advice is to use ASUS current stock firmware on the nodes

Is already on latest stock firmware. 386

Rovasteen, 2023-04-11:
Testing on TUF-AX5400.

WORKING:
DNS over TLS
IPv6 native
Guest network #2 bands 2.4 and 5.0 GHz
Primary network bands 2.4 and 5.0 GHz
SpeedTest
Logs normal
CPU load normal
RAM consumed is 55%, need a few days to see how much this grows
Network Tools -> Site Survey seems to work really well, a good improvement from Merlin

OBSERVATION:
Network Tools -> WIFI Radar still non-functional (not an issue with Gnuton build, common to all Asus builds)

michel130, 2023-04-12:
> hi @gnuton I'd like a try as well with the alpha firmware for the ZenWiFi_XT8 (base model: RT-AX95Q). Thanks for your efforts !

Hi,

I have also an ZenWiFi_XT8  and I used a couple of software for ZenWiFi_XT8 and the issue what I always have is the following:

My AiMesh is connected Uplink Type Wireless - 5GHz and everything is green.

After a day or sometimes longer it goes on 2.4Ghz and the line is orange.
the only way to get it back green is to reboot the 2 ZenWiFi_XT8.

I don't get it why it goes on 2.4Ghz, 

My 5GHz what says it is green has RSSi = -74 dBm

Now I have the original asus software on, model 1.0 and facing the same issues.

Do you have this also? or are you connected with a cable.

I hope the next firmware will stable what does not effect this issue.

dacootmeister, 2023-04-12:
Testing Model	ZenWiFi_XT8 (base model: RT-AX95Q)
Firmware Version	388.2_beta1 
I have some findings (all solved btw), so operating nicely. 

After installing on the main node the CPU utilization high for 3-5 minutes (AImesh nodes flashed afterwards without issue)
Known bug/issue: CPU core4 is always 0 %
[image]

DDNS failed registering; FIX: placed DDNS on disabled, re-enabled after a minute and re-registered my hostname -> OK

Memory utilization OK, with several AMTM packages installed/active
Total	498.06 MB
Free	102.63 MB

Will keep watch on behaviour and report @gnuton

micron888, 2023-04-12:
cfg_server issue show up again after 20hrs, even with fixed ip for aimesh

neil1962, 2023-04-13:
> OBSERVATION:
> Network Tools -> WIFI Radar still non-functional (not an issue with Gnuton build, common to all Asus builds)

Wifi Radar is working for me, are you giving it start time from the configure option/start/stop data collection:

[image]
[image]

Rovasteen, 2023-04-14:
RAM usage is at 56% after almost 5 days.   That is pretty good compared to pre-GPL 22525 which would have been 62% or more by now.

Rovasteen, 2023-04-14:
@neil1962 thanks for your post. That actually does work for the WIFI radar.

spysiuk, 2023-04-15:
Hi,
I would also like to test the alpha. I've got two RT-AX82U and one 92U. I'm on 386 because of connectivity issues on 388.

DiMENZi0N, 2023-04-16:
Well it appears nobody is testing the RT-AXE95Q yet. I have 2x and would like to test the alpha. Currently I have the master as 388.1 gnuton and the slave mesh node as stock 386. Had a re-occurring issue with slave node 6ghz dropping and master would lock up when slave was also 388.1 gnuton.

WarrenSchultz, 2023-04-16:
I have a AX-11000 running 388.2 merlin, I can put the alpha on the connected RT-AX92Us. Is there an issue with having a mix of 388.1 and .2 on the 92Us, or should I upgrade them all?

CPngN, 2023-04-16:
I have 2 ET8s, both on 388.1 for the past several days.  I've managed to
get my phone on 6ghz again for the first time since I bought them. But more
importantly, we've played games and steamed video to all corners of the
house and I haven't had to reboot the routers physically yet like I used
to!  I factory wiped and manually set up everything again sans some kid
filtering and I just might be able to survive with these routers!
Also a month ago I moved to wired backhaul using some 2gbps TPlink PLC
adapters to turn the house wiring into a network. That has worked wonders
also for the gaming machines.
Looking forward to final 388.2!

DervilRus, 2023-04-17:
Everything is fine on the AX5400. But there are a few places in the UI where elements are partially black (like TUF) but should be gray.

gnuton, 2023-04-19:
@DervilRus thanks for checking. feel free to open tickets for those bugs found otherwise I may skip it.
I am still spending time with TUF-AX3000 which seems to have been forked by ASUS and I have not got what's the reason for it.
It looks to me only V1 has been forked. Hopefully this is temporary.

hyprsnpr, 2023-04-20:
Testing with the rt-ax95q, been running all day so far on 388.2 and all is well as of now.

rterblanche, 2023-04-21:
So far no issues with 388.2 beta 1 on rt-ax92u. Been running stable but currently I do have a very simple install with 3 wired mesh nodes.

shiang-luong, 2023-04-21:
RT-AX95Q: this beta is working great for me along with the official firmware for the mesh nodes (3.0.0.4.388_23012).  Thank you!!!!

shiang-luong, 2023-04-21:
> Hi,
>
> I have also an ZenWiFi_XT8 and I used a couple of software for ZenWiFi_XT8 and the issue what I always have is the following:
>
> My AiMesh is connected Uplink Type Wireless - 5GHz and everything [...]

I had similar issues and even never getting 5GHz between the main and the node.  I hard resetted my main and my node and everything is working great.  Did you try doing a hard reset?

shiang-luong, 2023-04-21:
> traffic analyzer is broken [image]

Did you check if it is enabled?  (top right), it does also mentioned it takes a hour for data to refresh too.  
[image]

micron888, 2023-04-22:
@shiang-luong there is not button to enable or disable it.. tuf5400

Rovasteen, 2023-04-22:
At 12.5 days uptime, RAM usage stable at 57%.  Big improvement when compared to earlier 388 based builds which have much debated memory leaks.  On earlier builds, with the same configuration I would be approaching 70% RAM usage by now.

michel130, 2023-04-22:
> I had similar issues and even never getting 5GHz between the main and the node. I hard resetted my main and my node and everything is working great. Did you try doing a hard reset?

Yes I did that, what I do now, and I notes it works a bit better not 100% perfect:
channel bandwitch default 80MHZ and since that it stays on green.
Sometimes it goes on orange but after a couple of minute it is green again.

shiang-luong, 2023-04-24:
> @shiang-luong there is not button to enable or disable it.. tuf5400

I think that's the problem, your ui didn't update correctly? Try restarting the router and node or  reapplying firmware?

hyprsnpr, 2023-04-24:
Odd issue happened this afternoon. Swap jumped from 0 to 1.5GB and stopped being able to look up DNS names. Rebooted the router and all went back to normal. Could of been a weird fluke but thought I would report it. Never had that happen before. Was running fine on 388.2 for about 5 days before that though.

[13 more replies]