gnuton/asuswrt-merlin.ng #367: TUF-AX3000 V2

mattjonesuk, 2023-04-12:
**Is your feature request related to a problem? Please describe.**
I would like a version that is compatible with the TUF-AX3000 V2 router please. 

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

gnuton, 2023-04-13:
Please vote for it in the page https://github.com/gnuton/asuswrt-merlin.ng/issues/140 
Thanks a lot

closed 2023-04-13 by gnuton

DevangerTM, 2023-04-30:
Please support TUF-AX3000 v2