gnuton/asuswrt-merlin.ng #368: Low DSL speed in DSL-AC68U

rzeperoni, 2023-04-14:
**Router Model Affected**
Models: DSL-AC68U

**Firmware Version Affected**
386.9_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
No idea

**Describe the bug**
My DSL connection is 1gbp/s, using original Asus software I am able to achieve that speed while Gnuton is limited only to max 400mb/s

**To Reproduce**
Have 1gbp/s internet connection

**Expected behavior**
Achieve max speed

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

gnuton, 2023-04-15:
Reset to factory and set the right profile if you are using DSL.
You should also update to a more up-to-date release

rzeperoni, 2023-04-16:
Factory reset did not help. This was first flash directly from oem
firmware. Is there newer version? I downloaded it two weeks ago from 386
branch for my DSL-AC68U

gnuton, 2023-04-17:
newer version won't help.. there is something not right in your config.
can you tell me which DSL config you have to achieve 1 Gbps?
it would be also cool to know the one which give you 400 Mbps.

35b which is the fastest option has a teoretical max speed of 300/100.
So I am trying to understand how you can achieve more than theoretical speeds with DSL.

rzeperoni, 2023-04-19:
Hi

I'm so so sorry about that bug - apparently my ISP had some troubles
recently and now everything works as expected, I'm easily reaching 950mb/s.
I wanted to emphasize - *This was my ISP issue, not firmware. *