gnuton/asuswrt-merlin.ng #369: Low VRam in DSL-AC68U after updating to 386.9-gnuton1

MHKalantarian, 2023-04-18:
**Router Model Affected**
Models: DSL-AC68U

**Firmware Version Affected**
Version : 386.9-gnuton1

**Is this bug present in upstream Merlin releases too?**
IDK

**Describe the bug**
After upgrading to latest version my default VRam usage is up to 200mb and getting low VRam error on top of the router admin page. I tried reset factory, removing custom scripts, erasing jffs partitions and even removing my jffs usb stick that contained the script.

**Expected behavior**
Before the update with jffs scripts running and vpn 24/7 my ram usage would be 150mb at max, is there anything changed in this new version that affects vram?

gnuton, 2023-04-19:
you can run top and check what's the process consuming memory.
can you tell me the error message?

korayal, 2023-05-04:
I have been having the same issue, so I tried this command shared [here](https://www.snbforums.com/threads/alpha-386-9_alpha1-for-wifi-5-802-11ac-models-gt-ac2900-rt-ac68u-and-rt-ac86u.82209/page-2#post-808409):

``` sh
for line in `nvram show | grep ^[^=]*=$ `; do var=${line%*=}; nvram unset $var; done; nvram commit
```

I'm not entirely sure whether that command broke anything, but the NVRAM usage dropped from *63909* to *47881*

I have saved the output of the `[nvram_after_reboot.txt](https://github.com/gnuton/asuswrt-merlin.ng/files/11396455/nvram_after_reboot.txt)` command, before and after executing it. 

then I did a reboot, which brought the NVRAM usage back up to *58000*. Here are the files:

[nvram_initial.txt](https://github.com/gnuton/asuswrt-merlin.ng/files/11396456/nvram_initial.txt)

[nvram_after_cleanup.txt](https://github.com/gnuton/asuswrt-merlin.ng/files/11396457/nvram_after_cleanup.txt)

[nvram_after_reboot.txt](https://github.com/gnuton/asuswrt-merlin.ng/files/11396458/nvram_after_reboot.txt)

MHKalantarian, 2023-05-04:
I reverted back to stock firmware because in addition to this, OpenVPN speed was less than half in the router. The default firmware is working better for now.

gnuton, 2023-05-06:
This 387a56b8f25a660ac14ff064ed5b9e1db7ea3f5a should fix the issue.

MHKalantarian, 2023-05-08:
@gnuton Can you make a release on 386.10 branch?

gnuton, 2023-05-08:
@MHKalantarian sure. just tagged 386.10_0-gnuton0_beta1. 
The image will be online in about 40 mins

gnuton, 2023-05-08:
pre-released is now available

MHKalantarian, 2023-06-24:
@gnuton The issue with ovpn still persists after a short while of VPN connection WAN disconnects without showing any error as you can see in the logs, tried with multiple VPN configs, here is the log:
```
Jun 24 15:12:49 openvpn-routing: Setting client 2 routing table's default route through the tunnel
Jun 24 15:12:49 openvpn-routing: Routing all traffic through ovpnc2
Jun 24 15:12:49 openvpn: Forcing all to use DNS server 8.8.8.8 (OpenVPN client 2 is set to Exclusive DNS mode)
Jun 24 15:12:49 ovpn-client2[5932]: Outgoing Data Channel: Cipher 'AES-128-GCM' initialized with 128 bit key
Jun 24 15:12:49 ovpn-client2[5932]: Incoming Data Channel: Cipher 'AES-128-GCM' initialized with 128 bit key
Jun 24 15:12:52 ovpn-client2[5932]: Initialization Sequence Completed
Jun 24 15:15:14 WAN_Connection: Fail to connect with some issues.
Jun 24 15:15:35 WAN_Connection: WAN was restored.
```

```
Jun 24 15:17:54 openvpn-routing: Setting client 2 routing table's default route through the tunnel
Jun 24 15:17:54 openvpn-routing: Routing all traffic through ovpnc2
Jun 24 15:17:54 openvpn: Forcing all to use DNS server 8.8.8.8 (OpenVPN client 2 is set to Exclusive DNS mode)
Jun 24 15:17:54 ovpn-client2[5932]: Outgoing Data Channel: Cipher 'AES-128-GCM' initialized with 128 bit key
Jun 24 15:17:54 ovpn-client2[5932]: Incoming Data Channel: Cipher 'AES-128-GCM' initialized with 128 bit key
Jun 24 15:17:56 ovpn-client2[5932]: Initialization Sequence Completed
Jun 24 15:19:04 WAN_Connection: Fail to connect with some issues.
Jun 24 15:19:33 WAN_Connection: WAN was restored.
```

gnuton, 2023-06-28:
@MHKalantarian to me it looks like the issue is related to the WAN (connection to internet) and not to VPN itself.
Basically 
Jun 24 15:19:04 WAN_Connection: Fail to connect with some issues. <--- no internet
Jun 24 15:19:33 WAN_Connection: WAN was restored. <-- internet is back

MHKalantarian, 2023-06-28:
@gnuton The problem is that this only happens after a few minutes of connecting a VPN, and it's not a coincidence. Every time I checked the WAN gave up after connecting to a VPN. I tried it with a phone network through USB and it's the same. So it's not only DSL.

gnuton, 2023-06-30:
> @gnuton The problem is that this only happens after a few minutes of connecting a VPN, and it's not a coincidence. Every time I checked the WAN gave up after connecting to a VPN. I tried it with a phone network through USB and it's the same. So it's not only DSL.

Are you able to collect more logs of what happens?

MHKalantarian, 2023-07-21:
The issue has been resolved due to disabling Network Monitoring > Auto Network Detection. For some reason, it detected no connection and dropped wan constantly after activating VPN.

closed 2023-07-21 by MHKalantarian