gnuton/asuswrt-merlin.ng #370: SNMP bricks Router RT-AX92U

Wyehoota, 2023-04-21:
**Router Model Affected**
Models: RT-AX92U (others not tested)

**Firmware Version Affected**
388.1

**Is this bug present in upstream Merlin releases too?**
Unknown

**Describe the bug**
If SNMP is enabled on RT-AX92U, the router immediately goes into an infinite reboot cycle. Router reset does not help
To recover you need to reflash factory firmware with ASUS recovery tool/

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Administration'
2. Click on 'SNMP'
3. Enable SNMP
4. Set any required options and Apply.
5. System immediately goes into infinite reboot cycle.

gnuton, 2023-04-21:
thanks for reporting. 
BTW If the problem is SNMP, factory resetting should disable it and the problem would go away.

Wyehoota, 2023-04-22:
Checked with 388.2beta1 and fault also exists.

Using reset switch to attempt factory reset does not work. Unit does end up in recovery mode after next reboot cycle
Factory reset via WEB is not available.

Reflash of gnuton/merlin firmware result in router still in reboot cycle.

To recover - flash FACTORY firmware, then do at least a recover of setting with SNMP disabled. Then able to reflash gnuton/merlin firmware.

Router in basic factory configuration with exception of IP address and syslog attached to rsyslog to catch log. No AiMESH.

Syslog dump from remote syslog server trap

Apr 22 12:33:44 RT-AX92U-31C0-D7B02E5-C rc_service: httpd 1266:notify_rc restart_snmpd
Apr 22 12:33:44 RT-AX92U-31C0-D7B02E5-C kernel: Unable to handle kernel NULL pointer dereference at virtual address 00000000
Apr 22 12:33:44 RT-AX92U-31C0-D7B02E5-C kernel: pgd = ffffffc013e9e000
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: [00000000] *pgd=000000000a167003, *pud=000000000a167003, *pmd=000000000a0c0003, *pte=0000000000000000
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: Internal error: Oops: 96000007 [#1] PREEMPT SMP
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: CPU: 1 PID: 3064 Comm: snmpd Tainted: P           O    4.1.51 #2
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: Hardware name: Broadcom-v8A (DT)
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: task: ffffffc013a7c1c0 ti: ffffffc010dc0000 task.ti: ffffffc010dc0000
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: PC is at ioctl_handle_mii+0x120/0x1b8 [bcm_enet]
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: LR is at ioctl_handle_mii+0x120/0x1b8 [bcm_enet]
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: pc : [<ffffffbffc426338>] lr : [<ffffffbffc426338>] pstate: 60000145
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: sp : ffffffc010dc3c20
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: x29: ffffffc010dc3c20 x28: ffffffc010dc0000 
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: x27: ffffffc000535000 x26: 0000000000000036 
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: x25: 0000000000000184 x24: 00000000fffbcf94 
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: x23: ffffffbffc429d80 x22: ffffffc010dc3d28 
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: x21: ffffffc015f75800 x20: ffffffc010dc3d28 
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: x19: 0000000000008947 x18: 0000000000000000 
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: x17: 0000000000000000 x16: ffffffc000186708 
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: x15: 0000000000000000 x14: 0000000000000000 
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: x13: 0000000000000000 x12: 0000000000000020 
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: x11: 0101010101010101 x10: fefefeff31677364 
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: x9 : 7f7f7f7f7f7f7f7f x8 : fefefeff31677364 
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: x7 : 0000000080808080 x6 : 0000000000000000 
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: x5 : ffffffc000846000 x4 : ffffffc000846460 
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: x3 : 0000000000000005 x2 : 000000000000ffff 
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: x1 : 00000000fffffff7 x0 : 0000000000000000 
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: Process snmpd (pid: 3064, stack limit = 0xffffffc010dc0020)
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: Modules linked in: nf_nat_sip init_addr(          (null) -           (null)), core_addr(ffffffbffd09c000 - ffffffbffd09d998)
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel:  nf_conntrack_sip init_addr(          (null) -           (null)), core_addr(ffffffbffd091000 - ffffffbffd093ff8)
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel:  nf_nat_h323 init_addr(          (null) -           (null)), core_addr(ffffffbffd08b000 - ffffffbffd08c0e8)
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel:  nf_conntrack_h323 init_addr(          (null) -           (null)), core_addr(ffffffbffd079000 - ffffffbffd07ca00)
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel:  nf_nat_rtsp init_addr(          (null) -           (null)), core_addr(ffffffbffd074000 - ffffffbffd074a00)
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel:  nf_conntrack_rtsp init_addr(          (null) -           (null)), core_addr(ffffffbffd06e000 - ffffffbffd06f110)
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel:  nf_nat_ftp init_addr(          (null) -           (null)), core_addr(ffffffbffd06a000 - ffffffbffd06a2c8)
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel:  nf_conntrack_ftp init_addr(          (null) -           (null)), core_addr(ffffffbffd064000 - ffffffbffd064c30)
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel:  thfsplus(O) init_addr(          (null) -           (null)), core_addr(ffffffbffd041000 - ffffffbffd04e494)
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel:  tntfs(PO) init_addr(          (null) -           (null)), core_addr(ffffffbffcfb4000 - ffffffbffd000488)
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel:  tfat(PO) init_addr(          (null) -           (null)), core_addr(ffffffbffcf5f000 - ffffffbffcf90368)
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel:  uas init_addr(          (null) -           (null)), core_addr(ffffffbffcf56000 - ffffffbffcf58138)
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel:  usb_storage init_addr(          (null) -           (null)), core_addr(ffffffbffcf3d000 - ffffffbffcf40c18)
[... cut ...]
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: [<ffffffc0003f3d00>] dev_ifsioc+0xa8/0x398
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: [<ffffffc0003f436c>] dev_ioctl+0x2c4/0x618
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: [<ffffffc0003c09ec>] sock_do_ioctl.constprop.17+0x4c/0x60
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: [<ffffffc0003c1890>] compat_sock_ioctl+0x628/0xf90
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: [<ffffffc0001867a4>] compat_SyS_ioctl+0x9c/0x15d8
Apr 22 12:33:45 RT-AX92U-31C0-D7B02E5-C kernel: Code: 6b02003f 54000060 51002421 94fe1627 (f9400001)

gnuton, 2023-04-23:
ok thanks for the explanation. This makes sense.
The only thing which still doesn't is the not working reset to factory button which should work.
I will check this. thakns for reporting.

gnuton, 2023-05-09:
@Wyehoota do you have chance to check the latest beta?

Wyehoota, 2023-05-10:
With
Date: April 28, 2023
Release: 388.2-gnuton0_beta1
results arrear to be the same. Log from click of "Apply" to system lockup attached.
[R.log](https://github.com/gnuton/asuswrt-merlin.ng/files/11446586/R.log)

gnuton, 2023-05-29:
SNMPD will be temporarely disabled for RT-AX92U (75272712f0cfe5c128e2ff0b6ba586c868c01c8b)