gnuton/asuswrt-merlin.ng #373: [DSL-AX86U][388.1] Connection down after a few days bug and No Internet with Manuel DNS

R3D-PLAN3T, 2023-04-24:
**Router Model Affected**
DSL-AX82U

**Firmware Version Affected**
388.1_0-gnuton1

**Is this bug present in upstream Merlin releases too?**
IDK

**Describe the bug**
when I manually select DNS for my router, I have no longer internet and WAN LED on router turns red. No matter what DNS address I take. If I take the DNS from my internet provider, I have internet.
I've had this problem since the last version.
2x Factory reset does not helped. Both time I didn't restore a backup. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to WAN
2. Click on Edit PVC
3. Scroll down to WAN DNS Setting
4. Select manuel DNS

**Screenshots**
[image]
[image]

rlfrank165, 2023-04-24:
> *Router Model Affected*
> Seen on both XT12 and XT8
>
> *Firmware Version Affected*
> 388.1_0-gnuton1 and also base ASUS 386 versions
>
> *Is this bug present in upstream Merlin releases too?*
> IDK
>
> *Describe the bug:*
>
Common ASUS failure mode on multiple versions

Sending this to this list actually in the hope that someone reading it
might have connections back to ASUS, as it's clear to me that this is
behind problems in multiple versions of both the ASUS original code and
Merlin versions based on it going back to various 386 releases.

A failure mode which appears as if the router has failed with
multiple versions have this in common:

nodes seem no longer able to connect to the Internet, with DNS failure to
reply at its core.

But if one can get to the router UI and one goes into Nework Tools and
tries to use the nslookup tool, everything appears to be fine.

If one can then get to one of the hosts that is failing, and does a
nslookup, it fails, but on this host the configured DNS address is the in
the router and not the real DNS server (i.e., the DNS address in active use
is the router address it got from DHCP, not the real DNS address).

If one gives (in this case Windows) nslookup a server command like SERVER
8.8.8.8 and then retries a lookup, it works fine and if one configures the
actual wifi interface to use the real DNS server address instead of the
router address ot also appears to start working again.

So the failure appears to be something corrupted in the router so that when
a node is using the router to forward a DNS lookup it fails, but if the
node is directly requesting a DNS lookup from the actual name service,
everything is continuing to work fine.

I have noticed this failure mode on everything from based 386
configurations to 388 based versions of both ASUS original code to Merlin
and Gnuton versions based on those Asus 388 versions.

Once this starts to happen rebooting fixes it (for a while) but once it
starts happen it then continues to happen even more frequently and the only
way to get things stable again is to do a factory reset and reconfigure the
router with a saved cfg file that predates the problem starting to occur
(i.e., a recent cfg file may already have corrupted information that causes
this problem to start occurring).  Not positive about this last point, but
the basic failure mode constantly repeats itself ever more frequently once
it starts occurring.

No idea if this information is helpful to anyone.

>
>

MagicStuffCL, 2023-04-27:
I have exactly the same problem in version 388, I use the latest version 386 and everything is stable

rlfrank165, 2023-04-27:
So what is weird is that I was seeing this failure mode on various both
Asus and Merlin 388 releases, and so went back to the Asus latest 386
release and it was working fine for about 2 months, and that all of a
sudden starting appearing even in the 386 release,  So something had gotten
corrupted (albeit slowly) even in the 386 releases, and was re-booting (at
first bi-weekly and then nightly), and then even multiple times a day.  So
once something gets corrupted clearly it's affecting the stored
configuration.  Last night I did a factory reset and re-configured from a
saved old config file (the one which has worked fine for several months) so
will see if we return to some stability.  But looks like whatever is
causing this still has roots going back to at least 386 releases.

Randy

NovalisVonHardenberg, 2023-05-08:
Yes, same DNS-Problem with stock firmware...

gnuton, 2023-05-08:
@NovalisVonHardenberg what's the stock version you tested?

NovalisVonHardenberg, 2023-05-09:
> @NovalisVonHardenberg what's the stock version you tested?

Latest, 3.0.0.4.388_22048

MagicStuffCL, 2023-05-28:
a few days ago I installed version 388.2_2-gnuton0_beta2 and the problem seems to be solved, this with two ax92 in aimesh mode, thanks for all the effort on the project