gnuton/asuswrt-merlin.ng #374: 388.2 beta1 on ET8/RT-AXE95Q - firmware update only shows main router and no remote mesh nodes.

CHF-UK, 2023-04-29:
Hi gnuton,

With 388.2 beta 1, the only issue i've found is only the main router is shown in the firmware upgrade screen, the 3 remote nodes dont show (1 wires, 2 wireless).

[image]

[image]

Other than that, this appears a great release.  Uptime currently just short of 24 hours and everything connected wirelessly seems to have a lot more reliable connection with clients no longer being randomly disconnected for minutes at a time.  Everything seems stable even if pushed relatively hard. 

[image]

bconrad0321, 2023-04-30:
I can confirm on my ET8 as well.

paul-hostetler, 2023-05-21:
This is also an issue in 388.2 beta 2. The firmware upgrade webpage only shows the main node. The iOS app does show both nodes.

gnuton, 2023-05-30:
fix in https://github.com/gnuton/asuswrt-merlin.ng/tree/BUG-374_merlin-function-missing-in-www-sysdep

gnuton, 2023-05-30:
tested and confirmed 4830eb669921b0b90b083467f7279b608c874ba7 fix the issue

closed 2023-05-30 by gnuton