gnuton/asuswrt-merlin.ng #375: Reboot Node ?

pstgh, 2023-04-29:
**Is your feature request related to a problem? Please describe.**
Not really a problem, just an enhancement to be able to reboot a specific node separately.

**Describe the solution you'd like**
Just a simple request to have an option to reboot a node device separately.

**Describe alternatives you've considered**
I can SSH into the node and reboot it that way, but I'd prefer not to always have SSH access enabled.

gnuton, 2023-04-30:
Hi,
I have not tested this but you can just mimic what the web page does
I have not tested this but I guess you can run something like this. 

`curl 'http://router.asus.com/apply.cgi' \
  --basic --user "$USERNAME:$PASSWORD"
  -H 'Content-Type: application/x-www-form-urlencoded' \
  -H 'Origin: http://router.asus.com' \
  -H 'Referer: http://router.asus.com/index.asp' \
  -H 'Upgrade-Insecure-Requests: 1' \
  -H 'User-Agent: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/112.0.0.0 Safari/537.36' \
  --data-raw 'action_mode=reboot&action_script=&action_wait=90' \
  --compressed \
  --insecure`

neil1962, 2023-04-30:
You can restart nodes individually from within the AiMesh menu or via the appNeil Morgan

pstgh, 2023-04-30:
Thanks!  You're right- if you go AiMesh> click on the node that you'd like to reboot> Management> Reboot Node
I appreciate the help!
*****
Philip Hager
Atlanta, Georgia