gnuton/asuswrt-merlin.ng #376: XT8 light shows white, but entire stopped working

brodly, 2023-04-29:
**Router Model Affected**
Models: XY8

**Firmware Version Affected**
Router- Gnuton 386.07
Node - 388_22525

**Is this bug present in upstream Merlin releases too?**
Not sure

**Describe the bug**
Access to route stopped working randomly, wasnt doing anything outside of normal usage. All devices lost connection via wifi, could not reach admin page, toggling wifi on various devices did not let me reconnect to wifi. I assumed the router would have rebooted itself, waiting about ~5min but still no access. Ended up manually turning off/on both router and node which got everything working again. The boot time did take a lot longer than usual, normally takes about ~1min to boot, but this time seemed to take around ~3-4mins.

Provided syslog 

**To Reproduce**
Steps to reproduce the behavior:
n/a, see bug report for info

**Expected behavior**
I would expect this not to happen :)

**Screenshots**
[syslog.txt](https://github.com/gnuton/asuswrt-merlin.ng/files/11359927/syslog.txt)

gnuton, 2023-04-30:
Hi @brodly 
Thanks for reporting this.
78 days of uptime is not bad.
In any case something went wrong in the wireless drivers.

I do see several:
kernel: CFG80211-ERROR) wl_cfg80211_sta_info : GET STA INFO failed, -21
but I cannot see a clear pattern or reason why that happened.

I have a 388_2 firmware update in the pipeline.
If you wanna give it a try.

brodly, 2023-04-30:
Thanks. Ill wait until 388.2 becomes more stable, but appreciate the review. Had a similar issue a while back where you reported the drivers causing issues so maybe some exposed memory leak after longer uptime

matchman72, 2023-05-08:
I experienced similar behavior when using Gnuton 388.2 or Asus firmware 3.0.0.4.388.22525. 
Then I flashed my XT8 device (master and nodes) with the latest Asus firmware 3.0.0.4.388.23012, the problem is resolved and my router is running stably for 21 days till now

gnuton, 2023-05-09:
388_2 for ZenWifi is based on 388_22582 so it could be that 3.0.0.4.388.23012 has solved the issue. 
Wireless drivers and AiMesh are closed source components. Those comes straight from ASUS and broadcom

gnuton, 2023-05-10:
@brodly I see that in the changelog of the latest release.
no plans at moment to move to next GPL in the current 388.2