gnuton/asuswrt-merlin.ng #378: 388.2 beta1 on ET8/RT-AXE95Q - 6G Is Missing From Smart Connect Settings

bconrad0321, 2023-04-30:
**Router Model Affected**
Models: (ET8)

**Firmware Version Affected**
388.2 beta 1

**Is this bug present in upstream Merlin releases too?**
Unsure

**Describe the bug**
6 Ghz is missing from band steering options in Smart Connect setup page on the ET8.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Smart Connect Settings

gnuton, 2023-05-09:
Confirmed and I have not found a solution for this.
i am looking for a fix,

Wyehoota, 2023-05-17:
If AiMesh is configured, I suspect that Smart Connect does not permit use of the "Backhaul Band" even if the backhaul band has been enabled as "backhaul and fronthaul".
In my modem AX92U if no AiMesh configured 
    then SmartConnect permits all 2.4G, 5G-1 and 5G-2 in Tri-Band, 
    or 5G-1 and 5G-2 in 5G Smart Connect Mode,
but if AiMesh configured SmartConnect only uses 2.4G and 5G and SmartConnect is really in Dual-Band mode not Tri-Band mode.