gnuton/asuswrt-merlin.ng #383: WiFi Radar

qqw1231, 2023-05-09:
WiFi Radar

There is a problem with the top here, and there is also a problem with the status bar

qqw1231, 2023-05-09:
> WiFi Radar
> 
> There is a problem with the top here, and there is also a problem with the status bar

tuf-ax3000  388

gnuton, 2023-05-09:
@qqw1231 which image the one with TUF UI?
any chance to show me a screenshot?

qqw1231, 2023-05-10:
[image]

qqw1231, 2023-05-10:
388.2

gnuton, 2023-05-11:
What's the problem? translation problem or what?

qqw1231, 2023-05-12:
> 有什么问题？翻译问题还是什么？寻求帮助和提示
> […转到辅助设置以更改键盘快捷键](#)

No top logo and left sidebar

gnuton, 2023-05-13:
Thanks for reporting I see the problem and it's affecting also other models.