gnuton/asuswrt-merlin.ng #386: 388.2 Beta 2: AX-5400 Low upload speed with Wireguard turned on.

Sp0Xik, 2023-05-12:
**Router Model Affected**
Models: AX-5400 v.1

**Firmware Version Affected**
388.2 beta 2

**Is this bug present in upstream Merlin releases too?**
Yes (388.1 beta 1)

**Describe the bug**
Upload speed with Wireguard enabled is still limited to 250 Mbps. Although the download speed after the update became as it should (750 Mbps). At the same time, on the Merlin forum, I saw that other people's speed, not only downloads, but also uploads, increased to the required values.

**To Reproduce**
Steps to reproduce the behavior:
1. Enable VPN or VPN Director using Wireguard.

**Expected behavior**
I expect the upload speed to be equal to the download speed.

**Screenshots**
[image]
[image]

gnuton, 2023-05-13:
@Sp0Xik did you tried to reset to factory?
can you show me a spedtest ookla speedtest?

BTW no need to create a new issue for it.. the previous one was fine too

Sp0Xik, 2023-05-14:
Hey @Gnuton
Here is a speed test with Wireguard enabled:
[image]

Here is a speed test without VPN:
[image]

After installing 388.2 beta 1, I reset the router to factory settings, but the problem was the same. I don't think that the dirty update on beta 2 could somehow affect this.

gnuton, 2023-05-14:
was wireshark in 388.1 performing better?

Sp0Xik, 2023-05-15:
No. in 388.1 when wireguard was running, NAT acceleration was disabled, from which both speeds rested at 250mbps