gnuton/asuswrt-merlin.ng #387: TUF-Ax5400 Traffic stats gui

micron888, 2023-05-14:
Unstable: 388.2_2-gnuton0_beta2
TUF AX-5400
GUI isue
[image]

gnuton, 2023-05-14:
What's the problem it doesn't show data or just broken UI?

micron888, 2023-05-14:
No on/off button
Cannot show data

Nehaxak, 2023-05-14:
Can confirm. Same problem here, same router, same latest beta.

Agreed to privacy. Same issue.
Withdrew from privacy. Same issue.

Fresh install, same issue.

Just tried a few of the above to test.

Just seems like a gui issue, gui button/switch missing to enable/disable.

gnuton, 2023-05-14:
have you noticed the same issue in the beta1?

micron888, 2023-05-14:
Same in beta 1

Nehaxak, 2023-05-14:
Issue was not there for me in the beta 1.

gnuton, 2023-05-29:
duplicate of https://github.com/gnuton/asuswrt-merlin.ng/issues/403

closed 2023-05-29 by gnuton