gnuton/asuswrt-merlin.ng #389: Node keeps disconnecting

ThonMayo, 2023-05-16:
I have the XT8 and it appears that the node keeps randomly disconnecting and reconnecting from the router on the latest beta 388.2.  I use the wireless backhaul. 

Connection between the node and the router was never an issue on previous releases. 

Both the router and node had a hard reset after installing the firmware. 

Looking forward to the stable release.

Schabeus, 2023-05-17:
I have the same problem with XT8 v1 (router plus 1 node). I moved from 386.07_2-gnuton1 which was rock stable to 388.2_2-gnuton0_beta2, did a factory reset and the wireless backhaul connection with the node became completely unstable. It keeps disconnecting continuously. I tried disabling 160mhz, manually setting channels, enabling/disabling DFS channels but it doesn't change anything. I had similar problems with Asus official firmwares from 388 branch. I'll try to test 388.1_0-gnuton1 later.

ThonMayo, 2023-05-17:
In general the 386 builds were really good. Had a few issues now and then but nothing game breaking. 

After moving to 388.1, my smart TV keeps dropping connection with the router. Hoped that 388.2 may fix that but didn’t, and now there’s the node dropping connection with the router randomly. 

I’m going to try the latest official ASUS firmware to see if it resolve those issues. I haven’t tried the official one since they moved from the 386 to 388 firmware. 

I’ll come and try the gnuton 388.2 when it’s in its final release.

gnuton, 2023-05-17:
@ThonMayo let me know if ASUS 388 original firmware works better, I would also interested if switching back to 386 would really fix the problem you are now experiencing.

SSkizz, 2023-05-18:
I've been poking around this issue for a couple hours now and it seems that there's a CPU capping and ram memory leak bug(?) that's hit a lot of ASUS routers today. PROD version is mostly what I've seen so far but its possible that's what's going on here as well.

gnuton, 2023-07-11:
If anyone wanna to try the first 388.4 pre-alphas please be my guest!
https://app.circleci.com/pipelines/github/gnuton/asuswrt-merlin.ng/1537/workflows/ccb6862d-a755-4398-b939-e580d0793679
You can download them straight from above link from the "Artifacts" tab.
Please note I have not tested all of them yet. So THEY CAN BRICK YOUR ROUTER or have BIG ISSUES.
But once again.. if you feel corageus feel free to test them and report back

firasdib, 2023-07-11:
@gnuton is there a changelog?

gnuton, 2023-07-11:
https://github.com/gnuton/asuswrt-merlin.ng/blob/DEV_388_23588/Changelog-NG.txt
This comes with updated GPL and drivers, and latest ASUS patches

Schabeus, 2024-02-12:
I dug up the thread to inform you that in my case the described problems disappeared after installing the latest official Asus firmware (3.0.0.4.388_24609). I've been using it for a week, and so far I haven't had a single case of node or SmartTV randomly disconnecting, which happened all the time on all previous 388 firmwares (official and gnuton). It seems, that Asus  have finally fixed this problem. Moreover, the WiFi speed has also improved. In the parts of the apartment where it was the weakest, the download speed increased by approximately 100 - 200 Mbps.