gnuton/asuswrt-merlin.ng #394: Wireguard Issue with Surfshark VPN

malcolmdavis224, 2023-05-21:
**Router Model Affected**
Models: RT-AX95Q

**Firmware Version Affected**
388.2_2-beta2

**Is this bug present in upstream Merlin releases too?**
Yes

**Describe the bug**
When using Wireguard functionality (with Surfshark VPN specifically), certain servers are either completely nonfunctional or only function for around 2-3 minutes before having no internet connection entirely. I have attempted to turn off NAT, change DNS settings, allow inbound firewall and change keepallow time but nothing has been effective in keeping the network working.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to "VPN"
2. Click on 'VPN Client'
3. Click on 'Wireguard'
4. Create Wireguard profile by uploading Surfshark Wireguard VPN config file to system
6. Turn on Wireguard
7. Click on VPN Director
8. Add 'WAN' to new rule

**Expected behavior**
Wireguard should connect to Surfshark VPN servers and provide a consistently working internet connection for the selected server in the Wireguard section.

gnuton, 2023-05-22:
was it working in the previous release?

malcolmdavis224, 2023-05-22:
It was not working in 388.1 stable either. Here is a syslog

Wireguard is turned on at the May 22 11:53:22 mark and turned off at 11:59:21. OpenVPN is also turned on during this time period 
[syslog.txt](https://github.com/gnuton/asuswrt-merlin.ng/files/11530866/syslog.txt)

malcolmdavis224, 2023-05-22:
A few seconds after a new pairwise key handshake is completed, pages stop loading. Every page is loading before that moment

LarryIsBetter, 2023-05-22:
I had something similar with Mullvad VPN where sometimes it won't complete a handshake after turning on the connection, I can tell because it says however many bytes sent but it'll always be 0 bytes received. IVPN works just fine I thought it could've been a Mullvad issue but after reading this maybe not? I can send Mullvad VPN Wireguard files for testing I use RT-AX82U btw Mullvad works just fine on everything else like laptop phone.

gnuton, 2023-05-27:
@LarryIsBetter are you running the very same config on both router and laptop?

LarryIsBetter, 2023-06-22:
@gnuton Sorry I've forgot to tell you but I've tested the stable for a while with Mullvad and the issue with away.

gnuton, 2023-06-22:
do you wanna keep this ticket open?