gnuton/asuswrt-merlin.ng #395: make ip addresses highlightable for copying

lsaadeh, 2023-05-22:
**Is your feature request related to a problem? Please describe.**
it would be helpful to be able to highlight and copy ip address

**Describe the solution you'd like**
make ip address highlightable 

**Describe alternatives you've considered**
manually typing ip addresses

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

DJREMIXED, 2023-05-22:
You can highlight in the web interface from network map tab and “view list” button. Is usually just export out what I have there sometimes. That way I can keep an export of the view list export file open while in other menus when I’m changing or configuring something like dhcp reservations or MAC address black/block lists. It’s part of Asus code I would figure and I find that most of the menus that have this limitation of highlighting on web data is because the data is generated my another service or piece of software and given to the httpd service. That is so you get to see a “view” of that current information until said service refreshes again. Prob why the info is a bit delayed in there. If I knew another way of getting the same information from the terminal ssh console for each I would just disable the web service completely. I can get most of that I want but would be curious if it would be done and presented in the console instead so I could indeed remote the web interface completely. I’m sure configuration will always in more cases be easier with a web front end but I’m more thinking about set it and forget it / log if services. Like a microwave by holding down the 2 button, I can disable the beep noise of my microwave. The. If I want the beep noise back I hold 2 down again and it comes back. Same with MacBook with disabling sleep so I can close my MacBook screen without having the laptop go to sleep and disconnecting all of my connections I was in. 
sudo pmset -a disablesleep 1
enable it again
sudo pmset -a disablesleep 0

[image]

gnuton, 2023-05-31:
since the above works there is no point in addng this feature.

closed 2023-05-31 by gnuton