gnuton/asuswrt-merlin.ng #396: iOS attached client list disappear when only primary node configured

phostetl, 2023-05-23:
**Router Model Affected**
Models: RT-AXE95Q

**Firmware Version Affected**
388.2_2 Gnuton Beta

**Is this bug present in upstream Merlin releases too?**
Unknown

**Describe the bug**
I have a two-node system.  I removed the secondary node and the list of connected clients in the iOS app was empty.   Connected clients did show in the web interface.

**To Reproduce**
Steps to reproduce the behavior:
1. Remove all nodes leaving only the primary node.
2. Open iOS app.
3. Devices list is empty.
4. See error

**Expected behavior**
The list should math the web interface.

**Screenshots**
If applicable, add screenshots to help explain your problem or remove this section

phostetl, 2023-05-23:
I found this because I was having some problems with the Wi-Fi connection periodically.  I suspected it might be a backhaul issue.  After I removed the other node I noticed the issue.  I readded the node and the problem disappeared.  I'm now running a wired backhaul which seems to have improved the Wi-Fi stability.

gnuton, 2023-05-31:
can I close this issue?

phostetl, 2023-06-01:
Please close the issue.

closed 2024-03-23 by phostetl