gnuton/asuswrt-merlin.ng #397: VPN protocol: Lightway support?

Jacky10X, 2023-05-24:
The WireGuard is already supported but the Lightway is not supported. 
And the Lightway is open-source.
So I suggest the firmware support the Lightway protocol.

https://www.expressvpn.com/blog/lightway-open-source-security-audit/

gnuton, 2023-05-25:
@Jacky10X have you checked if it's available in the entware repository?
Or does it require additional kernel support?

Jacky10X, 2023-05-25:
@gnuton 
it does not available in the entware repository.  
it seems it does not require additional kernel support.
https://lightway.com/docs/lightway-core/1.0.0/faq.html#is-lightway-in-the-linux-kernel-like-wireguard

gnuton, 2023-05-25:
unlikely something is gonna be added to the router images without extensive testing.

Jacky10X, 2023-05-25:
ExpressVPN has its own specific firmware for their own router Aircove and some routers including ASUS routers. 
https://www.expressvpn.com/vpn-software/vpn-router#check-your-router-model

Their firmware supports Lightway protocol. So i think this also suitable for firmware?